shadow-1/yi-hack-v3 #138: Any way to adjust the motion sensitivity when using yi-hack firmware?

ajkessel, 2018-02-02:
My camera looks out onto a yard and thus is triggered almost constantly by small motion -- trees moving slightly with the wind, for example. Is there any way with the yi-hack firmware installed to modify the threshold for what counts as "motion" triggering a saved file?

kamkelsey, 2018-02-17:
Within the yi app there are are options for this.  Depending on your camera/firmware there may be options for sensitivity and setting a zone within the frame that gets checked for motion.  This firmware should not effect the functionality of the yi app.

ajkessel, 2018-02-19:
I haven't been able to get the Yi App to communicate with the camera. Apparently I need the Chinese app but I don't believe that can be downloaded from the U.S. App Store. Is there any workaround for that?

kamkelsey, 2018-02-20:
From the readme:

“iPhone users will need to change their App Store to the Chinese App Store and install Chinese version of the Yi Home app. Link: https://itunes.apple.com/cn/app/xiao-yi-zhi-neng-she-xiang-ji/id931168730 “

mikeschi, 2018-02-20:
Here are some instructions I found and use for changing from the US to China Apple app store.
https://www.dropbox.com/s/nhx680fo0ukn1dx/HOWTO%20change%20App%20Store.docx?dl=0