shadow-1/yi-hack-v3 #146: Store video on NAS rather than Yi Cloud

zwiebelspaetzle, 2018-02-09:
Can this firmware be used to store video on my NAS rather than Yi Cloud?  Perhaps with a script that polls for new files and copies them via FTP?

jjrojoc, 2018-02-11:
Please, search, there are another's issues talking exactly same thing

kamkelsey, 2018-02-17:
No.  Not at this time anyway.

RTSP functions are not available yet per several other issues.  More information can be found by reading through the other issues and the readme file.

frekel, 2018-02-19:
See: https://github.com/shadow-1/yi-hack-v3/issues/79#issuecomment-348161594

tanmays, 2018-11-30:
If anyone is interested I made this script to regularly poll for new videos and copy them to the NAS. (the script runs as a cron job on the NAS)

https://gist.github.com/tanmays/09210334644cf8e0516e5ed4f30bd8ed