shadow-1/yi-hack-v3 #147: Unrecoverable Dome Camera 1080P with solid yellow light - fried?

losbenitos, 2018-02-11:
Ok so I flashed my EU1080P Dome model YHS.2016 a few weeks back with the hackv3 root and home firmware and all went smoothly, I could access the web menu and the camera was functioning perfectly for a week or so.

Sometime later after the camera was power cycled a few times, it dropped off the network and on boot displayed only 4-5 rapid blue flashes followed by a solid yellow colour.

I downloaded the recovery images home_h20 and rootfs_h20 which when placed on a fat32 1GB card do result in a sequence of 4 yellow flashes, followed by another 28 or so, but the camera then returns to a solid yellow light.

Experimenting with cycling hackv3 files and recovery files the camera recognises the changes each time and pulls the files off the card but no combination result in the camera functioning.

I notice the firmware provided by Yi through[ these instructions ](https://help.yitechnology.com/hc/en-us/articles/230776087-Manual-Firmware-Update-Instructions), the home file is named home_h20**m**, however feeding this to the camera does nothing with or without the 'm'.

It seems as though some other part of the file system has become corrupt as both the root and home updates apply without issue but ultimately it's failing to boot at all.

While it is a mystery what caused the camera to crash, I would quite like to get it functioning again, are there any other partition files available I can feed it?

losbenitos, 2018-09-06:
While I've had no suggestions on what to do with this bricked camera, I've taken another step and got my soldering iron out and got this connected up to PuTTY in the hope I can see something, however on powering up all I get is:

**▒**

**System startup**

I was hoping for a bit more than this, especially since it appears some LED activity takes place when switching between hacked and recovery copies of home_h20 & rootfs_h20, with or without an SD card inserted I get the same result.

Simon-46, 2020-10-17:
Good evening. I was wondering if you ever got this problem resolved. I have had the same thing happen after less than a day of use. Camera was paired and functioning as should, untill it was turned off for a few hours. Hasn't worked again since. Combinations of solid and beeping yellow light with the camera rotating every now and then. Ive attempted everything that can be googled. Every version of everything related. Very frustrated.

losbenitos, 2020-10-18:
Hi Simon

I feel your frustration. Alas no, I reached the end of the road with the Yi camera and eventually replaced it with a Wansview Q3S which supports the protocols I need right out of the box.

I'd be interested to know if you find out any way to restore the Yi as I still have it somewhere. Good luck