shadow-1/yi-hack-v3 #149: Yi Home Camera 1080p Solid Yellow Light

OmegaNalphA, 2018-02-15:
I am using the Yi Home Camera 1080p with rootfs_y20 and home_y20, I have put my SD card into the slot, but all that happens is a solid yellow light. Anybody else with the same issue?

OmegaNalphA, 2018-02-15:
Whoops was using home_h20 accidentally, checking with home_y20 instead on an older firmware version

OmegaNalphA, 2018-02-15:
Nope, still a solid yellow light. I am using a 32GB Samsung Micro SD, I just copied over the 2 files(rootfs_y20 and home_y20) to the root of the SD card and plugged it in. It doesn't seem to be flashing the firmware.

dardob, 2020-10-19:
Any luck with solving the problem? I'm also trying to change firmware in Yi Home 1080p,and got exaclty the same issue. MicroSD 32GB, formatted in FAT32, copied rootfs_y20 and home_y20 to direclty to root folder and... nothing. Once I turn on camera, it simply   works, but fw is not being updated. Got a stable yellow light from the very beggining till camera reach operational mode (blue light)

OmegaNalphA, 2020-10-22:
no clue, sorry dont think I ever resolved this issue

closed 2020-10-30 by OmegaNalphA