shadow-1/yi-hack-v3 #150: Fragmented Video Record in Cloud Service

junmin, 2018-02-17:
As attached image, we can see video record are fragmented in the cloud service. Does somebody know what's going on? thank you.

[image]

kamkelsey, 2018-02-17:
It looks like possibly your camera is set to record only when sensing motion and the sensitivity is set to high?
That is what mine was doing until I made an adjustment in settings.

junmin, 2018-02-17:
@kamkelsey I have been checking around about the settings you mentioned, but I cannot find it. Can you tell me where I can change it?

kamkelsey, 2018-02-17:
@junmin

Can you tell me what version of the camera you have?

Also, it may be that the cloud service only records video when there is motion, which would make sense in terms of bandwidth/storage use.  

I don’t use the cloud service myself but saw this when recording to sd card.  I should have mentioned that.

junmin, 2018-02-17:
@kamkelsey 

It's a Yi 1080p Home

kamkelsey, 2018-02-17:
Ok, I have two of these cameras.  I use the iOS app but it should be a similar process on Android.  The pc app does not seem to offer these settings.

On the main screen click on the settings button below the camera preview:

[image]

Then click on settings under “smart detection”:

[image]

That should bring up this page where the adjustments can be made:

[image]

If this does not work I’m afraid I won’t be of any help as I have iOS/macOS devices only and cannot check other options.  I use the latest yi firmware in conjunction with shadow’s hack.