shadow-1/yi-hack-v3 #158: Yi 1080p Home - Time zone change

Malexoff, 2018-02-28:
Could you, please, help to change time zone in camera?
I added GMT+5 in /etc/TZ. Timestamp on video is OK, names of folders is correct (according actual time), but rewind in Yi Ants (Yi Home) android app is not working.
Maybe you can add settings in Web Interface for change time zone.
Thank you

imrgill, 2019-02-11:
I will create /etc/TZ file and found same issue. timestamp on video ok but can not replay video.

shadow-1, 2019-02-20:
@Malexoff @imrgill 
This is a known issue with the Yi Home app.
It is easy to change the timezone on the camera using the `/etc/TZ` file. However it completely messes up the Yi Home smartphone app.

I have not tried it on a EU or US version of the camera. However with the CN version of the camera, the timezone is not adjustable in the Yi Home app and changing the timezone on the camera messes up with playback.

Since this firmware does not work without the Yi Home app. I did not implement the ability to change; time, date, timezone within the web interface.