shadow-1/yi-hack-v3 #167: Yi 1080p dome, model no. YYS.1917, failed to be flashed witrh yi-hack-v3

brother003, 2018-03-16:
I bought a china model yys.1917. I could not use it outside china. I tried to flash it with yi-hack-v3 files rootfs_h20 and home_h20 with different micro-sd cards. The flashing process did not begin at startup. It seems that the model is different from the other 1080p dome. Can anyone figure out if I am correct? Thanks very much.

tommillski, 2018-04-06:
Having same issue. Have tried 10+ times, resetting camera, reformatting 32gb SanDisk and adding correct files on. Doesn't start flashing process - goes straight to 'waiting to connect'.

I have the YHS.2016 model.

Radio-van, 2018-04-19:
Hi. I had trouble with flashing my Yi 1080p Dome too. Solved it by renaming yi-hack-v3 files from "home_h20.dms" and "rootfs_h20.dms" to "home_h20" and "rootfs_h20". Don't know why downloaded files have .dms ending, but seems that they don't need it: after renaming everything works.
Don't know if it's suitable for your situation, but maybe you have same issue.