shadow-1/yi-hack-v3 #169: Camera Offline Due To Proxy Server Down

huazzz, 2018-03-23:
Guys need help.
Been trying other proxies from website provided here but cant find a working proxy.
I can receive motion alerts but apps detect my cam is OFFLINE.
Any working proxies for me to try? or is there other issues?

Sookw, 2018-03-23:
U may refer to below issue, the problem still don't have any solution yet...

all yi dome camera offline??
#93 opened on Dec 8, 2017 by fredyagu

huazzz, 2018-03-23:
But mine is the 1080p camera which is working perfectly and nearly a year already.. until this morning it says camera offline at xxxxx... but i am still able to view it on my pc and also receive motion notifications on my phhone...

huazzz, 2018-03-25:
I just setup an opened spare Yi Home 1080p cam and its working.
As for my old ones, i tried everything but still shows OFFLINE AT : xxxxxx...
Is there anywhere we can hide our serial numbers ? as i suspect they have blocked us through serial numbers. I changed new Micro SD, still same.. but the new one is still kicking...

huazzz, 2018-03-27:
Anyone? admins?