shadow-1/yi-hack-v3 #170: Can I control camera on/off via telnet/ssh using this hack?

shades0425, 2018-03-24:
I want to switch camera recording without using the smartphone application. Is it possible to control on / off of camera recording via telnet/ssh using this hack?

bambiman, 2018-08-16:
I would like to know this too

mcgurdan, 2018-11-01:
I think I’ve found how to do this. Going to test today, but it’ll involve a couple of shell scripts (from a pi for me).

I’ll then have an api for Google (via IFTTT) that’ll just run the shell script (camera on or camera off) for each of my camera

mcgurdan, 2018-11-02:
Looking at the processes on the camera - the easiest way to disable the camera and re-enable the camera is as follows. The follow commands are done via ssh or telnet - they can be configured in a bash script and run remotely. Ideally I want this configured with flask, however, there appears to be a limitation to this. If I run the shell script interactively it works just fine, but with Flask it doesn't appear to run properly.

Disable:
1) killall watch_process
2) killall p2p_tnp

Enable:
1)  /home/app/watch_process & 

This process will ensure all the correct processes are running and gracefully bring back the system into a good state.