shadow-1/yi-hack-v3 #171: can't connect to remote host freevpn.ninja?

williamwl1215, 2018-03-25:
wget -O /tmp/temp_proxy_list.txt https://freevpn.ninja/free-proxy/txt 

Connecting to freevpn.ninja (202.71.99.195:443)
wget: can't connect to remote host (202.71.99.195): Connection refused

Return Code: 1

is there something wrong with freevpn.ninja website?   fail to get the proxy list and i even can not open this webpage no matter how i try. 

or may you change or add other ways to get or install proxy list for us to hack?  because i almost get into the last step and failed. bit frustrated..haha

Au8ust, 2018-03-25:
Same issue here, is there any workaround?

Au8ust, 2018-03-25:
@williamwl1215 you may check this https://github.com/shadow-1/yi-hack-v3/issues/135