shadow-1/yi-hack-v3 #172: Yi 1080p Dome Offline even after pairing

Robin2019, 2018-03-25:
After new firmware reset and proxy configuration, it's successfully paired between mobile app and the camera. In the beginning the mobile app works even fine to control the camera and monitor the room, but after a while it's always showing "Offline at: 2018/03/17 23:35" and the app can't connect to the camera again. 

Have tried to reset the camera a few times, and every time both mobile app and camera can be paired successfully. But just a while, same offline error appears again.

Killuaxx, 2018-03-27:
Same issue here

lovex0710, 2018-03-28:
Same issue on my Yi 1080p Dome.
Anyone know how to fix it? 
Thanks.

Dp4610yz, 2018-03-29:
i am also having the same issue....need to close the app and reopen...before can view...but once i exit the view for the camera....it goes offline....is there any work around???

Lennyfoo, 2018-03-31:
I have the same problem. Any solution?

RaymondCKL, 2018-04-01:
I also facing the same problem, need assist. Thanks.

It's always showing "Offline at: 2018/03/18 19:24"

loganlim78, 2018-04-17:
i also...it happen only when u disconnect and connect again. camera are working and only can not connect to our phone. i know is working cause i put in motion mode and it follow when you move around. i try so many way like format memory card, install new YI apps also the same.

vip991988, 2018-04-19:
I have the same problem, when i reboot or unplug the electric, i cant connect to camera

vip991988, 2018-04-22:
Anyone can fix this issue?

vip991988, 2018-04-22:
i have the same problem