shadow-1/yi-hack-v3 #176: Could not connect to freevpn.ninja

szefkamil, 2018-04-07:
Connecting to freevpn.ninja (185.53.179.7:443)
wget: TLS error from peer (alert code 40): handshake failure
wget: error getting response: Connection reset by peer

Return Code: 1

I have this problem when i want to downlad proxy list because few days ago camera is offline.

adrianwy, 2018-04-10:
I have the same issue. Could you please advice what we can do?

adrianwy, 2018-04-10:
I have found the https://github.com/shadow-1/yi-hack-v3/issues/135 but non of them are working

szefkamil, 2018-04-10:
http://spys.one/proxys/CN/

go to this website and tap manualy

random_chain
chain_len = 1
tcp_read_time_out 15000
tcp_connect_time_out 8000
[ProxyList]
socks5 223.241.214.44 1217
socks5 211.95.255.42 1080
socks5 118.89.231.84 30687
socks5 119.29.163.54 1080
socks5 61.145.213.14 1080
socks5 218.58.210.233 1080
socks5 221.209.62.170 1080

in system config sheet

[image]

szefkamil, 2018-04-10:
[image]

please use this and tap manualy socks5 and ip server please remeber about dots between numbers for example [socks5 211.95.255.42 1080]

adrianwy, 2018-04-10:
Yes that is the same link. I have just put all the Proxy on the configuration file and tested but non of them are working :(

szefkamil, 2018-04-10:
please try few of them maybe this one

221.232.233.159:1080 | SOCKS5

szefkamil, 2018-04-10:
after that please reboot camera by 
[image]

szefkamil, 2018-04-10:
and try to use app in smartphone and everything should be okay

szefkamil, 2018-04-10:
wait for blue led inform that wifi connection is okay in chinese

szefkamil, 2018-04-10:
remove other proxy leave only this one

adrianwy, 2018-04-10:
Ok will do and I will let you know, thanks Kamil

szefkamil, 2018-04-10:
no problem, i want to help you
i spend last sunday morning on this and have a lot problems too

szefkamil, 2018-04-10:
please sent me a picture of your website and i will check
and we wil try to resolve it together

szefkamil, 2018-04-10:
random_chain

chain_len = 1

tcp_read_time_out 15000
tcp_connect_time_out 8000

[ProxyList]
socks5 218.72.111.83 1080
socks5 114.67.67.67 1080
socks5 183.159.92.171 1080
socks5 117.64.50.173 1080

szefkamil, 2018-04-10:
this is my setup

szefkamil, 2018-04-10:
any progress??

adrianwy, 2018-04-10:
Did you check the 'Check Proxy List'? Does it work for you? Non of them are working

szefkamil, 2018-04-10:
no I didn't check this

szefkamil, 2018-04-10:
only write proxy from this web and reboot camera

szefkamil, 2018-04-10:
put phone to scan qr code and wait for wifi connection
and click next on the app

adrianwy, 2018-04-10:
Which version do you have? Chinese or international?

szefkamil, 2018-04-10:
chinese

razorj, 2018-04-10:
I have the same issue, which was solved by repeatedly trying more Proxy a few days ago. However, I just found the problem occurs again, so I would assume the proxy is not valid anymore. Is there solution can last longer? How to make a automated proxy list just like old service by freevpn.ninja. 

By the way, mine is also Chinese version. Thx a lot!

szefkamil, 2018-04-11:
Hello all,
Yesterday i check this metod and it's working fine:

1. download app from google play store: kibou VPN 
2. copy few proxy adress to proxychains-ng tab
3. reboot camera
4. camera is working and connecting.

Reagards
Kamil

szefkamil, 2018-04-11:
other method is to use vpn spys.one/free-proxy-list/CN
copy few adresses with ratio near to 100% and it's working.

I checked also

adrianwy, 2018-04-11:
Ok thanks Kamil for your support. I will try to manage it this evening

szefkamil, 2018-04-12:
> Thanks Kamil it's working ;)

My pleasure

williamwl1215, 2018-04-13:
sorry i still couldnt connected the proxy server, no matter how many proxy server address which are 100% uptime i copy,  when test config server connectivity still failed.  reboot can not solve the problem either. 

is there anything wrong with my config? 

such as last time i test the proxy 
HTTP 101.200.56.177:3128	
SOCKS5 120.79.78.178:1080	
SOCKS5 114.99.70.111:1080	
 and still always show me like this
[image]

szefkamil, 2018-04-13:
Remove : before port example
Socks5 192.192.192 1080

nelson2mts, 2018-07-12:
szefkamil still cannot connect sametime can sametime cannot pls help me tq.

strelokr, 2019-02-08:
new list
socks5 149.129.107.164 1080
socks5 118.193.190.35 8088
socks5 103.115.44.150 1080 
socks5 192.151.246.90 3128

tomnyw, 2019-03-14:
Hi there 
I have tried all the proxy servers from http://spys.one/free-proxy-list/CN/ and i cant seem to connect. 
is there any way to go about this ?

szefkamil, 2019-03-24:
Fuck this camera and we should buy a new unlocked in US or EU