shadow-1/yi-hack-v3 #179: anything wrong with server connection why i test failed.

williamwl1215, 2018-04-13:
hey your manual type server has all failed.
and i try to type other servers and all test config failed.

is there any problem or mistake i made? like type the wrong server address formate?  couldnt connect
[image]
 the proxy server, no matter how many proxy server address which are 100% uptime i copy, when test config server connectivity still failed. reboot can not solve the problem either.

is there anything wrong with my config?

such as last time i test the proxy
HTTP 101.200.56.177:3128
SOCKS5 120.79.78.178:1080
SOCKS5 114.99.70.111:1080
and still always show me like this

williamwl1215, 2018-04-13:
[image]
[image]

Log
wget -O- -q -T 8 https://ipinfo.io 

{
  "ip": "42.153.49.122",
  "city": "Kuala Lumpur",
  "region": "Kuala Lumpur",
  "country": "MY",
  "loc": "3.1667,101.7000",
  "postal": "50586",
  "org": "AS38322 WEBE DIGITAL SDN. BHD."
}

Return Code: 0

proxychains4 wget -O- -q -T 8 https://api.xiaoyi.com 

[proxychains] config file found: /home/yi-hack-v3/etc/proxychains.conf
[proxychains] preloading /home/yi-hack-v3/lib/libproxychains4.so
error: invalid item in proxylist section: 	

Return Code: 1

proxychains4 wget -O- -q -T 8 https://api.xiaoyi.com 

[proxychains] config file found: /home/yi-hack-v3/etc/proxychains.conf
[proxychains] preloading /home/yi-hack-v3/lib/libproxychains4.so
error: invalid item in proxylist section: 	

Return Code: 1

proxychains4 wget -O- -q -T 8 https://api.xiaoyi.com 

[proxychains] config file found: /home/yi-hack-v3/etc/proxychains.conf
[proxychains] preloading /home/yi-hack-v3/lib/libproxychains4.so
error: invalid item in proxylist section: 	

Return Code: 1

anyone can understand what is my problem?  Thanks so much boss

williamwl1215, 2018-04-13:
socks5 223.241.214.44 1217
SOCKS5 120.79.78.178:1080	
SOCKS5 114.99.70.111:1080
HTTPS 119.28.26.57 3128                
HTTP 119.28.50.37 82    
HTTP 119.28.112.130 8000    
HTTPS 119.28.138.104 3128  
HTTP 119.28.37.58 80             
HTTP 118.25.45.94 1080                  
HTTPS 119.28.22.236 3128    
HTTP 140.143.96.216 80
SOCKS4 61.234.37.3 1080                
SOCKS4 61.234.37.118 1080  
SOCKS5 193.112.88.59 1080  
SOCKS4 111.230.171.66 1080
HTTP 114.250.25.19 80           
HTTP 122.72.18.60 80             
HTTP 122.72.18.35 80             
SOCKS5 222.128.117.99 1080
SOCKS5 123.58.249.65 1080
HTTP 122.72.108.53 80           
HTTP 193.112.1.24 3128                  
HTTP 119.27.177.169 80         
SOCKS5 116.196.100.186 9000       
HTTP 61.4.184.180 3128                  
HTTP 14.152.84.112 3128                
SOCKS5 139.199.230.242 1080
HTTP 119.28.221.28 8088                
HTTP 111.231.253.19 3128             
SOCKS5    123.207.172.101 9999   
SOCKS5    123.206.187.209 9999             
HTTP 119.28.99.194 3128                
HTTPS 106.48.49.6  80

Those are all the proxy server i have typed, I found and type from variaties free vpn website, all tested should be work, but why all those are failed....

lywme, 2018-04-16:
I have the same problem, don't know how to solve it.

chqzeng, 2019-03-11:
please use lowercase, such as "socks5" instead of "SOCKS5".

sam1273, 2021-04-22:
I'm having problems that it show 
Error: invalid item in proxylist section:
And

Avichiii, 2021-10-18:
so do I? can any one help me with it?

Callmewhenyouwant, 2021-11-01:
You can to just change number of ip address and port that system told you it is the error item.if it still not work just change in to another number.I can help you just this cuz I tried it myself and it really work

ps.use proxychains in normal acc not root acc is one of my advice cuz some engine doesnt support root acc about proxychains

mix50, 2022-01-06:
You can check the correct writing by reading what's described into the .conf file. I had the same error and saw that's is told in my proxychains.conf

124Aniket, 2022-01-26:
use only http proxies , i also had same problem but now it is solved

AtypicalSysAdmin, 2022-05-08:
There should be no : between port and ip, just space. 

i.e. Valid format:
type ip port
socks5 47.91.149.178 8443