shadow-1/yi-hack-v3 #182: Do I need to downgrade to 2.1.0.0A and how?

wolfit50, 2018-04-18:
I can flash the hack firmware, I just bought the camera, and the current firmware is 2.1.0.0C_201803081641.

Do I need to downgrade to 2.1.0.0A and how?

gali8, 2018-04-18:
Look at the main page, at the end of the page there is the section "Going Back to Stock Firmware".

gali8, 2018-04-19:
Why do you need to downgrade??

wolfit50, 2018-04-20:
I could not flash the hacked firmware, I did every step at the main page, put home_y20 and rootfs_y20 on a formated Fat32 TF card, insert and plug power, it just didn't flash.

batrarobin, 2018-04-20:
@wolfit50
Maybe try a different card of lower capacity. 
Also ensure the filenames don't have any extensions when you copy them.
Only place both files on root path of the card and nothing else.

batrarobin, 2018-04-20:
Also try a different computer to copy files into the card.

wolfit50, 2018-05-07:
@batrarobin tried 4g/16g/32g cards, still can't flash the firmware.

xcray, 2018-05-31:
I've just hacked yi 1080p home with the newest official firmware, 2.1.0.0C_201803161855, everything works fine.
no filename extensions, fat32, nothing else.

edgarfebres, 2021-07-02:
I just bought 3 Yi 1080p home cameras, I've tried Yi-hack-v3, v4 and others, but nothing seems to be able to flash the firmware anymore. I double checked that the format of the SD card is FAT32, since I'm doing it from a Mac, and also checked that there were no extensions to the two files I place in the root of the card to no avail. Any recent incidents you guys might know about?