shadow-1/yi-hack-v3 #186: Error starting FFmpeg.

bsowens, 2018-04-26:
I just setup the firmware on my Yi Dome (not 1080p, international version). Here is the config for it:
```
camera:
  - platform: yi
    name: Camera
    host: 'xx.x.x.xxx'
    password: my_password
```

However I keep getting these two Errors:
```
Error starting FFmpeg.
```
and
```
FFmpeg fails program arguments must be a bytes or text string, not bool

Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.5/site-packages/haffmpeg/core.py", line 114, in open
    stderr=stderr
  File "/usr/lib/python3.5/asyncio/subprocess.py", line 212, in create_subprocess_exec
    stderr=stderr, **kwds)
  File "/usr/lib/python3.5/asyncio/base_events.py", line 1180, in subprocess_exec
    % type(arg).__name__)
TypeError: program arguments must be a bytes or text string, not bool

```

There are several forum posts online and I have spent hours and hours trying to debug this. Any adivce? I'm happy to provide more info if needed.

dmitrybabeshko, 2018-05-18:
#129 This helped me to resolve the problem with FFMPEG

bachya, 2018-05-28:
@bsowens This isn't the proper project to post this issue. Home Assistant-related issues should get posted in the Home Assistant project's issues page.