shadow-1/yi-hack-v3 #193: when Click "Download Proxies"

ClericLe, 2018-06-06:
It shows "ERROR: Could not connect to freevpn.ninja, ensure that you are connectected to the Internet." when Click "Download Proxies" at setp 16.
My internet is ok to use anything else.
Anyone has meet this before?

chrissong85, 2018-06-11:
Yup I'm facing exactly the same issue

tingsin86, 2018-10-12:
at the main page paste this below and save. and test
 you try proxy CN only , update at http://spys.one/proxys/CN/
Ex
socks5 122.4.161.117 1080
socks5 183.158.142.54 1080
socks5 183.6.168.196 1080
socks5 221.209.62.170 1080
socks5 61.145.213.14: 1080
socks5 117.135.132.107 3000
socks5 125.79.199.78 8888
socks5 120.133.6.6  9065
socks5 121.196.207.36: 1080
socks5 124.163.236.110 1080