shadow-1/yi-hack-v3 #201: Bricked camera, solid yellow light, help?

pomtom44, 2018-06-29:
Hey everyone.
Not sure if this is the best place to ask, but worth a shot
My friend had a Yi Home camera, and was messing around with the files on it, Not sure what he was doing, but somehow he bricked the camera.
He's given it to me as he doesnt know enough about them and can't be bothered fixing it, so i said id have a look and give it a go.

When booting up the yellow LED stays lit and nothing else happens, 
I have tried loading the home_y18 and rooty_18 files on, from various repo's
and just booting without the SD card, but nothing seems to work

At a guess I think he may have modified one of the core linux files and has bricked the internal OS

Is there any way to reload the OS onto the camera?
Or is it only good as a paperweight now?
Thanks in advance

xcray, 2018-07-04:
ttl, backup from same model by dd command, then restore by dd also.
That’s all

pomtom44, 2018-07-04:
Ah
Didn't know they had that
okay ill have a look and see what I can do and will inform back in a few days

pomtom44, 2018-07-20:
Finally got around to it, but now I dont know where to go.
Got a serial connection, booted the camera, and got a kernel panic.

Heres the whole serial log file.

`U-Boot 2010.06 (Aug 13 2015 - 16:32:20)

Serial Number: hi3518c

Check spi flash controller v350... Found
Spi(cs1) ID: 0xC8 0x40 0x18 0xC8 0x40 0x18
Spi(cs1): Block:64KB Chip:16MB Name:"GD25Q128"
MMC:   MMC FLASH INIT: No card on slot!
*** using default environment for 16MB flash
In:    serial
Out:   serial
Err:   serial

 do_auto_sd_update: detect flash size: 0x1000000, gsdexist=0

 do_auto_flash_update: detect flash size: 0x1000000, gsdexist=0

 do_auto_update: detect flash size: 0x1000000
USB:   scanning bus for devices... 2 USB Device(s) found
0 Storage Device(s) found
gpio7_7 =80
gpio2_5 =20
Press Ctrl+C to stop autoboot
16384 KiB hi_sfc at 0:0 is now current device

## Booting kernel from Legacy Image at 82000000 ...
   Image Name:   Linux-3.0.8-7518-hi3518-kernel
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2894932 Bytes = 2.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Linux version 3.0.8 (rock07@Server) (gcc version 4.4.1 (Hisilicon_v100(gcc4.4-290+uclibc_0.9.32.1+eabi+linuxpthread)) ) #1 Wed Apr 30 16:56:49 CST 2014
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
CPU: VIVT data cache, VIVT instruction cache
Machine: hi3518
Memory policy: ECC disabled, Data cache writeback
AXI bus clock 200000000.
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 10160
Kernel command line: mem=40M console=ttyAMA0,115200 root=/dev/mtdblock4 rootfstype=jffs2 mtdparts=hi_sfc:256k(boot)ro,128k(env),128k(conf),3072k(os),3584k(rootfs),9088k(home),128k(vd)
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 40MB = 40MB total
Memory: 35080k/35080k available, 5880k reserved, 0K highmem
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
    DMA     : 0xffc00000 - 0xffe00000   (   2 MB)
    vmalloc : 0xc3000000 - 0xfe000000   ( 944 MB)
    lowmem  : 0xc0000000 - 0xc2800000   (  40 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .init : 0xc0008000 - 0xc0029000   ( 132 kB)
      .text : 0xc0029000 - 0xc0515000   (5040 kB)
      .data : 0xc0516000 - 0xc053a0e0   ( 145 kB)
       .bss : 0xc053a104 - 0xc0557380   ( 117 kB)
SLUB: Genslabs=13, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:32 nr_irqs:32 32
sched_clock: 32 bits at 100MHz, resolution 10ns, wraps every 42949ms
Console: colour dummy device 80x30
[... 96 lines cut ...]
NET: Registered protocol family 10
NET: Registered protocol family 17
NET: Registered protocol family 15
lib80211: common routines for IEEE802.11 drivers
Registering the dns_resolver key type
registered taskstats version 1
drivers/rtc/hctosys.c: unable to open rtc device (rtc0)
usb 1-1: new high speed USB device number 2 using hiusb-ehci
VFS: Mounted root (jffs2 filesystem) on device 31:4.
Freeing init memory: 132K
/sbin/init: '/lib/libc.so.0' is not an ELF file
/sbin/init: '/lib/libc.so.0' is not an ELF file
/sbin/init: can't load library 'libc.so.0'
Kernel panic - not syncing: Attempted to kill init!
Backtrace:
[<c0033ff4>] (dump_backtrace+0x0/0x10c) from [<c04057f0>] (dump_stack+0x18/0x1c)
 r6:00000009 r5:c053a3f0 r4:c053a3f0 r3:0000000a
[<c04057d8>] (dump_stack+0x0/0x1c) from [<c0405858>] (panic+0x64/0x190)
[<c04057f4>] (panic+0x0/0x190) from [<c00456e8>] (do_exit+0x640/0x700)
 r3:60000013 r2:c2024000 r1:c2022120 r0:c04ba2a4
 r7:c2022000
[<c00450a8>] (do_exit+0x0/0x700) from [<c00458a8>] (sys_exit+0x18/0x1c)
 r7:00000001
[<c0045890>] (sys_exit+0x0/0x1c) from [<c0030320>] (ret_fast_syscall+0x0/0x2c)
`

pomtom44, 2018-07-21:
Nevermind Seem to have got it to work
Managed to load some firmware on through the sd card, no idea why it wasnt working before, but watching though the serial port I could see it loading the default files back onto the device.

Apapou, 2019-05-30:
Hi guys, 

I have the same bricked after update camera problem. I found this tread but as I am not an expert is there any chance to prompt what exactly to do to make it work again? I am afraid it is not clear to me how to connect is to computer, where to get the firmware (anyone can share please?), what is dd command etc? Will really appreciate your help.

Thanks,
Alex

Apapou, 2019-11-17:
No one replied but I have found a solution so leaving it here in case someone needs it. See these pages on official UI website. It works!

https://help.yitechnology.com/hc/en-us/articles/230776087-Manual-Firmware-Update-Instructions