shadow-1/yi-hack-v3 #205: Yi Home Camera 47USxxx on 1.8.7.0A_201702081101

Mkmgr, 2018-07-07:
Hi everyone,

I tried to flash my **Yi Home 720P / 47USxxx / YHS-113-IR** (**Europe version** : http://amzn.eu/9jXhWfC) running on **Yi OS 1.8.7.0A_201702081101** with the files "**rootfs_y18**" & "**home_y18**" available from the release "**yi-hack-v3 0.1.5**" and it didn't work :

_The yellow light flashes slowly ~20 sec, then the blue flashes slowly ~10 sec, then the shutter opens and finally the cycle starts over again. The camera never invites me to pair with the phone. I'm using the iOS Yi Home app on the european market._

I managed after many tries to restore the original OS but I would like to find a solution to avoid sending videos on the Chinese cloud.

Does anyone have the complete solution to apply the hack with this specific version of the camera?

Thank you in advance for your help!

mfizz1, 2018-07-10:
I have successfully applied the hack to 47US camera. I did connect to wifi before i flashed with this hack. I cant remember if i had to connect to wifi again.

Mkmgr, 2018-07-12:
Do you remember the firmware version of your Yi 47US before flashing them?

mfizz1, 2018-07-19:
Unfortunately no, I don't see any reason why it wouldn't work my camera is 2018/04 production date. Might be worth flashing the recovery first and then trying the hack

mfizz1, 2018-07-26:
Couple of further observations, this hack doesn't stop videos from being sent to the cloud. In fact the best part about this hack is the ability to use the app whilst still have root access to do other things. Secondly i would try with another card fat32 formatted with only these files on. What you're describing i have experienced but only when updating via the app. The fact  your camera is 47US means you can apply the hack

jaxdao, 2018-09-17:
you need to reset by pressing the RST button after flashing the yi hack firmware

michael11123, 2019-01-23:
@Mkmgr did you fix this errror?