shadow-1/yi-hack-v3 #215: recovery image issues

DanyHubGit, 2018-07-24:
I have installed the Firmware files required for my Yi Home 47US camera, but now I want to reinstall the original firmware.
I have tried to install the recovery image  Yi Home 27US / 47US (as suggested in the main page of repo), copying rootfs_y18 and home_y18 in sd, and restarting the camera, but the firmware hack-v3 doesn't disappear.
I have made also some test with Yi Home image, but with the same result.
Now I don't know how install the stock firmware.
Can you help me or suggest me the right way to proceed? thanks

DanyHubGit, 2018-07-25:
If I want to restore recovey image by FTP, is there any way to do it? tnxs

PercentageCorporation, 2018-08-25:
I had the same problem. I finally went to the Yi site and downloaded the firmware.
https://www.yitechnology.com/firmware/index/class/home
The files were version D:
home
home_y18m
I put them on the SD card and they loaded and I was able to connect.

TDPFlorida, 2018-11-02:
Is there a way I could load some stock software on my Yi Home 1080 camera and it would REMOVE the hack completely?
I've noticed when I flash the camera with different firmware (or the upgrade on the camera) if DOES NOT remove the hack?
Thanks

TDPFlorida, 2018-11-02:
I figured out how.
Thanks

Z-pi, 2018-11-08:
How did you do it?

TDPFlorida, 2018-11-08:
I will explain it in better detail later today when I  have time after work but until I understood, it was very confusing.

TDPFlorida, 2018-11-09:
Understand the process of upgrading (changing) firmware is TOTALLY different than installing/removing the hack firmware.

1. When changing STOCK firmware, you put 1 file on the SDCard root with the following naming conventions:
YI Home Camera 		“home_y18m” 
YI 1080P Home Camera 2	“home.bin”
YI Dome Camera		“home_v200m”
YI 1080P Dome Camera		“home_h20m”
YI 1080P Home Camera		“home_y20m”

2. Installing the hack requires 2 files on the SDCard root with the following naming conventions:
https://github.com/shadow-1/yi-hack-v3/releases
YI Home Camera 		“home_y18” 
YI 1080P Home Camera 2	“home.bin”
YI Dome Camera		“home_v200”
YI 1080P Dome Camera		“home_h20”
YI 1080P Home Camera		“home_y20”

3. To REMOVE the hack, requires 2 files on the SDCard root with the following naming conventions:
https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr
Yi Home | rootfs | home | Stock firmware is version   1.8.7.0D_201708091510.
-- | -- | -- | --
Yi Home 17CN | rootfs_y18 | home_y18 | Stock firmware is version   1.8.7.0C_201705091058.
Yi Home 27US / 47US | rootfs_y18 | home_y18 | Stock firmware is version   1.8.7.0A_201702081101.
Yi 1080p Home | rootfs_y20 | home_y20 | Stock firmware is version   2.0.0.1A_201612051401.
Yi Dome | rootfs_v201 | home_v201 | Stock firmware is version   1.9.1.0F_201701041701.
Yi 1080p Dome | rootfs_h20 | home_h20 | Stock firmware is version   1.9.2.0C_201611011902.
after removing the HACK, the firmware is returned to the stock firmware noted above and there is NO trace of the hack on the camera.

In each case, you put the file or files on the root of the SDCard, put the SDCard in the camera and cycle power on the camera. You will know if the firmware is being changed because when the camera powers on the yellow/amber LED will blink or stay solid for 30+ seconds.

I hope this helps.
Tom

Z-pi, 2018-11-09:
Thank you!, i am going to try it

Mikekayy, 2019-02-25:
Thank you @TDPFlorida for the instructions! Saved me from losing my sanity.

ccomincini, 2020-06-13:
Thankyou @TDPFlorida great guide