shadow-1/yi-hack-v3 #216: Applying Hack on 1080p 48us

mfizz1, 2018-07-25:
I have the 48US version of the camera. i had firmware 2.1.0C initially. I flashed the firmware succesfully and it was working perfectly with the app

I came home and the camera wasn't working so I unplugged and restart and it was stuck in a loop. I then applied the recovery firmware and it finally allowed me to connect. I then re applied the hack, which worked perfectly again but I couldn't connect the app anymore. I reset the camera and tried to connect it to the app a fresh and now the cameera won't pair. 

Now I am in a situation where it connects on stock and it is connected to wifi when hacked but it won't connect to the app. I upgraded to 2.1.0D and flashed and still same issue. If only i could find the 2.1.0C firmware perhaps i could test it and see what the difference is.

@shadow-1 how can I help you to fix this issue

aztazt, 2018-09-28:
I just bought two 48US "Yi Home 1080p v1" cameras, but the hack is not working on them. They are running on a 2.0.0.1A_201612051401 firmware and the original Yi Home android App is working perfectly like that.
I've formatted my SD card in fat32, changed settings in yi-hack-1080p.cfg and wpa_supplicant.conf, but the LED becomes fix blue a few seconds after the restart. nmap shows no opened port.
I have 4 Yi Ant cameras (720p) and all are working fine with the original Fritz-smg hack.
I don't have a TTL cable, what can I try next ? Could a firmware downgrade help ?
Thank you for your great job anyway.

morphirr, 2019-01-31:
I have this camera 48US I also had problems I applied the hack twice and also formatted the SD twice after the hack the firmware was on **2.1.0.0C_201710091352I** I also had problems it would get to the pairing stage then say time out and i had to reset camera so I plugged it directly into router for power and to be as close to the router as possible for the pairing, when I checked I could login to web server so i disabled Proxychains-ng and then did the pairing again seemed slow but finally paired and is working so far only did it a few hours ago, hope same doesnt happen to me @mfizz1  How can you tell which version camera you have @aztazt ?

morphirr, 2019-02-01:
@mfizz1 are you a mod as I see you have problems with firmware etc yet you have advised me that I should be able to update to latest firmware?? Are these patches suitable for camera 48US as it only shows 47US is there any difference between these two cameras and would this patch or updating to latest firmware cause a brick? I'm still on **2.1.0.0C_201710091352** uncertain what I should do as it took several tries to get the patch installed and all is working great at the moment love the ftp & web login thanks to all for their hard work, can't wait to see what else might be developed