shadow-1/yi-hack-v3 #218: yi-1080p-Dome does not pair after FW upgrade

Rhapsody02, 2018-08-02:
After firmware upgrade and reset of the camera, I re-start pairing process and go through all steps successfully up to connecting to wifi (also success), however then.... NOTHING and the pairing times out.

Any ideas, please?

Rhapsody02, 2018-08-15:
No reaction? :-/

One more thing I found out, the camera web interface is accessible after failed pairing.
If I cannot pair with yi-hack-v3, any idea how can I flash back the original firmware, please?

It does not begin flashing... Am not sure which file(s) and name(s) to use (tried multiple.

lvnilesh, 2018-08-17:
Did you skip step 11 in https://github.com/shadow-1/yi-hack-v3#getting-started---step-by-step-guide

Rhapsody02, 2018-08-17:
I guess that I deserved that, I missed I HAVE to disable the proxy settings, not optional.
I feel silly, but still it is great that it works now!

**MANY** thanks for your help!!!

closed 2018-08-17 by Rhapsody02