shadow-1/yi-hack-v3 #219: Compabile Firmware for new YI-Cloud DOME 1080P

eladiogalvez, 2018-08-05:
Can you confirm if will be available a firmware version for new YI-Cloud Dome 1080P.

Current official firmware is: 1.9.3.0B_201802051941

I will try to find chipset.

GoldenTK, 2018-08-28:
Same issue, there is hack-v3 for those cameras ?

MrBLJ, 2018-09-20:
Same for me ! Support would be very useful !

phoxy84, 2018-09-21:
Same here. Bought this camera and appears the non-international version. The hack does not flash and when installing the camera with the (chinese version of the) app the camera shuts down with the message that the camera only works in China. Hope there is a cleaver person who can fix this issue.

hunterelliott, 2018-09-23:
Would also appreciate firmware for the cloud dome 1080P!

totalitarian, 2019-07-10:
Did a firmware get released?

eladiogalvez, 2019-07-10:
I don’t tested but possible this project can help:

https://github.com/TheCrypt0/yi-hack-v4

phoxy84, 2019-07-11:
I can confirm that the Yi Hack v4 works on the cloud dome 1080p