shadow-1/yi-hack-v3 #228: Yi 1080p Outdoor EUR support

fedyfausto, 2018-09-08:
Hello guys! I have a Yi 1080p outdoor EUR version and i'm wondering if i can help you to support this camera, can i give to you some informations?

nonick321, 2018-09-08:
There is already a thread (#82). Unfortunately rtsp is not working at the moment..

fedyfausto, 2018-09-08:
yes i seen that thread and i flashed the firmware y20 reneaming to h30. The camera flashed the firmware but now the led still in yellow color

aelg305, 2018-09-08:
@fedyfausto I saw your comment on the other post about rtsp, any success getting there?

fedyfausto, 2018-09-08:
unfortunately not yet but i do not that i'm the best guy for resolve the problem :D

aelg305, 2018-09-08:
Cool. Anything I can do to help?

fedyfausto, 2018-09-09:
Mmmm i am not really a rtsp expert so i do not know any technologies or software to run a rtsp server for this cam, somebody can help me in some way?

fedyfausto, 2018-09-10:
guys i did it!
[image]

aelg305, 2018-09-10:
@fedyfausto did it...yaiiiii.... Now, somebody help this dude to get that code out. Lol

fedyfausto, 2018-09-10:
I'm have some torubles with rtsp because after some time che client crashs :S i'm using a server built by a user @andy2301 and this binary wants some of particular settings that i'm trying to discover what are the best values. If you help me i can upload the fix and the server bin / src :0

aelg305, 2018-09-11:
@fedyfausto 
Dummy here.... so:
-format the SD card
-drop the zip file (unziped) to the root of the SD card?
-turn on and if everything works, then turn off and move the files?
if not....step 2 from your post....right?

not working for me :(
Do i need to telnet or ssh into it?

fedyfausto, 2018-09-11:
wait you can not login with ssh or telnet?

fedyfausto, 2018-09-11:
> @fedyfausto
> Dummy here.... so:
> -format the SD card
> -drop the zip file (unziped) to the root of the SD card?
> -turn on and if everything works, then turn off and move the files?
> if not....step 2 from your post....right?
> 
> not working for me :(
> Do i need to telnet or ssh into it?

aelg305, 2018-09-12:
not letting me SSH or telnet! :( 

I followed this steps:
-format the SD card (on mac)
-drop the zip file (unzipped) to the root of the SD card..all files...
-then turned the camera ON with the SD card and everything with the Yi app (no telnet)
-then turn off and moved the 3 files
-still no telnet but yi app still works! :o

aelg305, 2018-09-12:
a sample of my supplicant...good??

`ctrl_interface=/var/run/wpa_supplicant
ctrl_interface_group=0
ap_scan=1

network={
	ssid=“name of SSID”
	key_mgmt=WPA-PSK
	pairwise=CCMP
	group=CCMP
	psk=“password_here”
	priority=2
}`

fedyfausto, 2018-09-13:
this is mine:
ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="[SSIDHERE]"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="[PASSHERE]"
}

aelg305, 2018-09-13:
Perfect, that helped.
I also read this https://www.freebsd.org/cgi/man.cgi?wpa_supplicant.conf(5) and it was great with the settings.... I hope it helps everyone else.

fedyfausto, 2018-09-13:
remember that you must use this only to insert the fix files and then you can restart the camera and use it like normally use but with telnet service and rtsp (needs to be fixed)

aelg305, 2018-09-13:
got telnet working, then took the sd out and had to create the folders home and then app /home/app and drop the 4 files into it, whats next?

fedyfausto, 2018-09-13:
No now if you can access via telnet you can list on /home/app directory and you can see a list of binaries of application. So if you list on /tmp/ad you can see the sd content. Now copy/replace that files of my fix into the /home/app and then you can test if all works resetting the camera (type reboot) and then configure it with application in normal way (press the reset button until you hear the voice from camera and then reset again or plug out , subsequently configure the camera with app)
From now you can try and test some value of RTSP editing the rtsp2303.sh and rebooting the camera.
That server does not work properly but you can find another bin of rtsp server and try it. Anyway I can upload the source code of rtsp2303 if you want

aelg305, 2018-09-14:
planning on comitting this to the git?

aelg305, 2018-09-14:
so, there was a **firmware update** which killed the hack.... so I went back:
FOR THOSE THAT NEED STEP BY STEP..like me... and need to replicate.

RSTP

-formatted the card
-downloaded the files from yi-hack (roof_y20) and renamed them to h30 and included my wpa_supplicant.conf and let it boot (yellow light comes one, then flashing yellow (meaning is flashing) and then solid yellow but no connection yet
-disconnected and drop the files from fedyfausto with the h30 and y30 and connected again
-yellow -> flashing -> solid -> connected after a couple of turn on/off from power
-telnet into it to confirm -> OK 
-copy  the following 4 files to the home directory using telnet (rtsp2302, rtsp2302.sh, libstdc++.so.6, init.sh)...used command: cp *.sh *.6 rtsp* /home/app
-sealed the deal with a power cycle (disconnect from wall/reconnect)

notes: 
-fedyfausto mentioned that we need to play around with the rtsp options and I havent been able to find one that works for me yet..(havent try any, but I will and report back)
-the Yi app does open and there is a stream, which is great!

Thanks to everyone...special thanks to fedyfausto....if there is someone that can figure out, its you! :)

nonick321, 2018-09-14:
Great work @fedyfausto.

 From time to time i get a stream in tinycam which breaks after some seconds. It reminds me to my xiaomi dafang. But if you set the stream Option to "fixedqp" the stream does not break anymore. 

I wonder if this Option exist in the source code of the rtsp2303?

Did you compile the source yourself? Perhaps wie could try to compile the rtsp Form the dafang hacks?

hiwing15, 2018-09-18:
Will this work on the Chinese version of the camera?

aelg305, 2018-09-20:
> > so, there was a **firmware update** which killed the hack.... so I went back:
> > FOR THOSE THAT NEED STEP BY STEP..like me... and need to replicate.
> > RSTP + Xiaomi outdoor camera
> > -formatted the card
> > -downloaded the files from yi-hack (roof_y20) and renamed them to h30 and included my wpa_supplicant.conf and let it boot (yellow light comes one, then flashing yellow (meaning is flashing) and then solid yellow but no connection yet
> > -disconnected and drop the files from fedyfausto with the h30 and y30 and connected again
> > -yellow -> flashing -> solid -> connected after a couple of turn on/off from power
> > -telnet into it to confirm -> OK
> > -copy the following 4 files to the home directory using telnet (rtsp2302, rtsp2302.sh, libstdc++.so.6, init.sh)...used command: cp *.sh _.6 rtsp_ /home/app
> > -sealed the deal with a power cycle (disconnect from wall/reconnect)
> > notes:
> > -fedyfausto mentioned that we need to play around with the rtsp options and I havent been able to find one that works for me yet..(havent try any, but I will and report back). I was able to get my camera software MotionEYE to recongnize the camera but image is not coming back yet... more to come.
> > -the Yi app does open and there is a stream, which is great!
> > -typing in my browser http://camera_IP or http://yi-hack-v3 does nothing, however, I do see the IP from the router and when scanned for open ports, only port 23 is open.
> > Thanks to everyone...special thanks to fedyfausto....if there is someone that can figure out, its you! :)
> 
> Thx guys, yes the next step (after rtsp) is that fix the dns and then create a web server/page from witch manage the camera

so, the extra files you have on your fix? or thats "coming soon"? :)

fedyfausto, 2018-09-21:
> Will this work on the Chinese version of the camera?
I think so, but i have not a Chinese Version, only EUR :(

@aelg305
Maybe when we have something that works with our cams

aelg305, 2018-10-03:
@fedyfausto 
Question, were you able to get any streams out? or does the stream keeps crashing?
I'm able to connect with telnet and the stream gets picked up by MotionEYE a very light weight simple software to monitor cameras....but NO actual image.

Willytoons, 2018-10-06:
@aelg305 mi cámara es la OUTDOOR YI 1080, el archivo no viene comprimido, y cuando lo descargas solo es uno y no 2 como en los otros modelos

Willytoons, 2018-10-06:
@aelg305 , estimado Andres, el nombre del archivo q se descarga para mi camara es 3.0.0.0D_201809111054home_h30m , sabrás por casualidad, que nombre le tengo que poner para conseguir la actualizacoión. Gracias.

Willytoons, 2018-10-07:
estimado @aelg305  casi todo lo del paso a paso lo puedo hacer, pero no tengo ni idea de como crear el archivo supliccant.conf  :(   :(

aelg305, 2018-10-08:
ok, el wpa_su varia dependiendo de tu configuracion.
Te toca leer un poco aqui: https://www.freebsd.org/cgi/man.cgi?wpa_supplicant.conf(5)
Mi configuracion del router usa WPA2-PSK [AES] y se conecta solo en el 2.4Ghz bandwidth

cambia los valores abajo para el ssid y el password... deja las comillas "..."

`ctrl_interface=/var/run/wpa_supplicant
ap_scan=1

network={
  ssid="nombre de la red"
  scan_ssid=1
  proto=WPA2
  key_mgmt=WPA-PSK
  pairwise=CCMP
  group=CCMP
  psk="password de la red"
}`

mikfaina, 2018-10-08:
English please

skywluke, 2018-10-10:
I try, but not clear for me, please help me: 
- 1st of all download 2 files (home_h30, rootfs_h30):
https://mega.nz/#F!s9wVzZQZ!k6F3p8HcWCoWl-jla2Cxpw 
- Put these files to SD card, and boot the camera connect to wifi, pair with phone.
- After disconnect the camera download this: https://mega.nz/#!9Fw0iYjZ!UmJ81DohmUpC1UetOW6JRvjR0ioHerm2qmh91XR1BGs
and then THIS TIME copy only the files [rtsp2302, rtsp2302.sh, libstdc++.so.6, init.sh] into your /home/app/ (what i have to create)

I made it, but nothing happend! No telnet, no ssh. :-( Please help me.

skywluke, 2018-10-12:
Dear Guys! @fedyfausto @aelg305 
First of all i installed to the camera newest official version. With filename home_h30m. 
I downloaded and renamed (from here https://github.com/shadow-1/yi-hack-v3/releases ) and put the sd card.
home_h30
home_y30
rootfs_h30
rootfs_y30

Downladed and replaced (on the sd card) @fedyfausto files home_h30 and rootfs_h30. 
Boot the camera. No telnet, no ssh. 
And then copy  the files [rtsp2302, rtsp2302.sh, libstdc++.so.6, init.sh] into /home/app/ (what i have to create) Result is same, no telnet, no ssh.
What i did wrong?

fedyfausto, 2018-10-12:
first of all preapre a wpa_supplicant file like this one:
ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="[SSIDHERE]"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="[PASSHERE]"
}

prepare your SD card:
1. format to fat32
2. put these files into it https://mega.nz/#F!s9wVzZQZ!k6F3p8HcWCoWl-jla2Cxpw
3. put the wpa_supplicanti nto the rrot of sd card

then, put the sdcard into your cam and then wait until the led stops to blink (so fixed yellow color)
After that, search on your local network the camera and test to telnet it

skywluke, 2018-10-13:
No, it doesn't start firmware update. Normal start, with original configuration.

aelg305, 2018-10-15:
> English please

English there:
https://github.com/shadow-1/yi-hack-v3/issues/228#issuecomment-421429340

skywluke, 2018-10-17:
Dear Guys! @fedyfausto @aelg305 ! Thanks, it's work! but now, the question is, how to use ftp server? Is it possible to use? Or how else to possible to download files from camera?

skywluke, 2018-10-17:
The answer for myself is:  busybox  tcpsvd -vE 0.0.0.0 21 busybox ftpd -w /tmp

Sygu00, 2018-10-22:
Hi!

@feyfausto @aelg305!! Thanks 4 the work!
I understand this works for Yi 1080p Outdoor but, does it for 27US/47US model? Is the same process?
I have two yi ANTS cam with yi-hack-v3 working but I need RTSP too.

Thanks in advance!

aelg305, 2018-11-06:
> first of all preapre a wpa_supplicant file like this one:
> ctrl_interface=/var/run/wpa_supplicant
> ap_scan=1
> network={
> ssid="[SSIDHERE]"
> scan_ssid=1
> proto=WPA RSN
> key_mgmt=WPA-PSK
> pairwise=CCMP TKIP [...]

any updates on getting the rtsp working?

arunsmu, 2018-11-11:
Hi i am having YI outdoor camera and i bought it from china. When i try to connect the camera with the wifi it's saying this camera can only be used within china. Now after seeing the above posts it appears that some have solved this issue successfully.  But the problem is i can't seem to understand lot of the technical stuff mentioned over here. It will be extremely helpful if someone can explain all the steps briefly step by step. It will be helpful if i can connect it to the wifi and make it record and also see it through live-stream.

akbaryahya, 2018-11-20:
any update this, about rtsp , how to get it?

rdiver, 2018-11-28:
> Dear Guys! @fedyfausto @aelg305 ! Thanks, it's work! but now, the question is, how to use ftp server? Is it possible to use? Or how else to possible to download files from camera?

Hi @skywluke how did you manage it in the end? I have been tearing my hair out for 2 days.

skywluke, 2018-11-29:
@rdiver : Telnet to camera. After that you have to start ftp server with the following command: "busybox  tcpsvd -vE 0.0.0.0 21 busybox ftpd -w / " Voila ftp is working. Now, if you would like to start ftp automaticaly with the camera boot, you have to find and open "/home/app/init.sh" file. Find "busybox telnetd &" Below this line put the ftp start command: "busybox  tcpsvd -vE 0.0.0.0 21 busybox ftpd -w / " Save & close & restart the camera and enjoy ftp server. That's all! ;)

rdiver, 2018-11-29:
> @rdiver : Telnet to camera. After that you have to start ftp server with the following command: "busybox tcpsvd -vE 0.0.0.0 21 busybox ftpd -w / " Voila ftp is working. Now, if you would like to [...]

Cheers @skywluke I'm more struggling with getting to that step. I have loaded multiple combinations of firmware. renamed rootfs_h20, rootfs_y20 to 30. All variations of ones above from the mega site.

My understanding was, download rootfs_y20 renamed to rootfs_h30 and home_y20 renamed to home_h30. Load onto camera, blinking yellow to solid yellow.
Then load on the files from the mega link above. Replace the 2 files already on the SD and add the wpa config. 
After doing this, I cannot get the camera to work. I actually have managed to brick the camera a couple of times. Then through what I can only call button bashing with the reset button and various firmware I managed to get it back to normal.

fedyfausto, 2018-11-29:
Hello guys! Forgive me for my disappear :D in the next days i'll try to fix the rtsp server (if i can), anyway in which way we could use the ftp to access to the stream?

quenrice, 2018-11-29:
Guys,

I followed the tasks mentioned and everything was working, I was on a 2017 firmware (out of the box)

And then, I wanted to try an upgrade on lastest firmware, (3.0.0.0D_201809111054)... Bad idea, it seems they counter the hack because now it doesn't work, I retry an upgrade from the rootfs_h30 and home_h30 but no telnet possible...

EpicLPer, 2018-11-29:
I just updated to the latest firmware, I hope this won't break anything...

mikfaina, 2018-11-30:
But  with the hack, I'm still able to view cameras from YI Home app?

arizzi, 2018-12-15:
@EpicLPer  I also have updated to  fw  3.0.0.0D_2018091110543.0.0.0D_201809111054 and it seems the hack is not working. Has anyone successfully hacked it with 3.0.0.0D ?

EpicLPer, 2018-12-16:
I haven't tried since people keep saying that the RTSP stream keeps crashing.

arizzi, 2018-12-16:
I managed, it seems to work also with 3.0.0.0D reverting it to B+mods.
I noticed that rtsp2301_watch.sh is actually a github webpage download instead of the actual file content that I guess was meant to be
https://github.com/xmflsct/yi-hack-1080p/blob/master/sd/test/app/rtsp2301_watch.sh

EpicLPer, 2018-12-16:
Lol what, seriously

gabrielegirardi, 2018-12-17:
Hi, I did the whole procedure and the camera works, the problem is that the stream rtsp: //192.168.0.13: 554 / ch0_0.h264 crashes very often, while both the app and the windows program work. Inside the FIX folder the files are not "rtsp2302, rtsp2302.sh" but "rtsp2303, rtsp2303.sh" go these well?

evanfloden, 2018-12-23:
I think the newer original firmware is stopping the yi-hack-v3 from working.

I am using model YHS.3017 and have tried most things. I've used this hack before on other cameras with no problem.

@gabrielegirardi  & arizzi Any ideas where we can find the  3.0.0.0B versions of the original firmware?

On the [download page](https://www.yitechnology.com/firmware/index/class/home) I only get 3.0.0.0D.

mvklingeren, 2018-12-30:
The unbrick guide for your: Yi Home Outdoor 1080p 👍 

@evanfloden i had the same issue, to unbrick a 3.0.0xyz.. device

To fix it, add these 2 files to a sdcard:https://mega.nz/#F!s9wVzZQZ!k6F3p8HcWCoWl-jla2Cxpw
the files in this zip are the 3.0.0B release, **this is the only way you gonna get it to work again**
(2.x files, like from the 'recovery' folder provided by shadow-1 do not work)

1) reboot cam, check if the lights blink (amber/blue) until its done, wait a min after it burns solid

2) then turn it off, take the card out, add a wpa_supplicant.conf and remove the other files from sd

keep it turned on from now, until last step

3) telnet into the device (use a network scanner to find the ip on your network if needed)
telnet: root, no password.

4) copy the content from this zip: 
https://mega.nz/#!9Fw0iYjZ!UmJ81DohmUpC1UetOW6JRvjR0ioHerm2qmh91XR1BGs 
(directory 'fix') to the sdcard. Put the sd back into the camera:

mount /dev/mmcblk0p1 /tmp/sd
cd /tmp/sd
cp rtsp2303.sh /home/app/
cp libstdc\+\+.so.6 /home/app/
cp init.sh /home/app/

remove the sdcard, and reboot the device (unplug, plug : )

happy camming with the Yi, big thanks to fredyfausto

(in case the links in this post stop working -> i got a backup, just let me know)

gabrielegirardi, 2018-12-30:
Thanks to all, I had already found this solution and the 5 cameras work well with both the App, PC and Home Assistant, the procedure you described is fine I guarantee, but if I do the update of the signature from 3.0.0B to the 3.0.0D, still work? has anyone tried?

mvklingeren, 2018-12-30:
Do not update to 3.0.0D, your setup will stop working.

And then you can fix it again with above guide ;-)

EpicLPer, 2018-12-30:
So is the RTSP stream working on the hack without any problems?

fedyfausto, 2018-12-30:
Unfortunately my fix will crash the rtsp client after some seconds... :( sorry guys i did not found solutions for now

[69 more replies]