shadow-1/yi-hack-v3 #231: 1080P home camera takes firmware but inst hacked?

eryk4381, 2018-10-17:
I have what I want to say is the newish 1080p home camera and tried the firmware hack and the light blink for about 30 seconds yellow and then blue and comes on. I can still access it from the app like nothing happened but SSH and the web interface are still not available. I tried taking it backing to the old recovery firmware and THEN flashing the hack firmware with the same results. What am I doing wrong?

morphirr, 2019-01-31:
this happened me too is your camera on **firmware 2.1.0.0C_201710091352** as thats what its on after its flashed, **Yi Home Camera 1080p 48US** I had to flash camera twice, formatted SD card twice, then it had problems pairing had to reset camera several times until I realized to try the web login and Proxychains-ng was enabled which I disabled and tried pairing again which worked

drlarsen77, 2019-02-02:
You always have to configure or disable Proxychains-ng, no matter which camera you have. If you look at Step #10 of the directions, it tells you as much. :)

neilsonwong, 2019-03-05:
> I had to do the same to get it working. Take it down to old firmware and then reflash. It then would work.

How did you take it down to the older firmware?
Is it just the yi-hack recovery images? or would you need to do it via another method?

eryk4381, 2019-03-05:
For some reason it took me a few times to get it flashed to the old firmware. I would try the sequence to flash the old firmware a few times with the auto reboots and see if it works. If not, you can try using a firmware that is in the middle of what you have and the yi-hack firmware to downgrade slowly.

shadow-1, 2019-03-05:
@neilsonwong @eryk4381 
Yes yi-hack recovery image is a good way to reflash the firmware or simply going to an older version of yi-hack-v3 before flashing to the latest version again.

Flashing the same version of yi-hack-v3 over itself does not always work. The cameras boot loader has a very primitive way of working out if the firmware upgrade file matches what is currently on the flash memory. If it thinks the firmware upgrade file matches what is already on the system, the firmware upgrade is skipped.

Almost always you have to flash to a different firmware before going back to the latest version to force the upgrade.