shadow-1/yi-hack-v3 #233: 1080P (4F CN) doesnt load the firmware

goterrier, 2018-11-04:
Tried loading home_y20 and rootfs_y20 on both formatted 32GB (sandisk) and 16Gb (samsung) SD card but seems like the camera is not loading the firmware at all. 

It just goes to bootup and waiting for connection.

I tried connect once and use a network scanner to detect the hardware. It detect the camera hardware manufacturer I had is by "Shenzhen Bilian Electronics Co., Ltd" and not under (“Shenzhen Zowee Technology Co. Ltd”).

Wondering different manufacturer will the way of loading the firmware be different? It seems like bypassing the firmware and not loading at all.

ykhandler, 2018-11-06:
It is not because different manufacturer, but it is because different chipset. 
You'll find a person in 4pda who able to flash the camera...

goterrier, 2018-11-06:
Seems tough to read Russian in 4pda.

ykhandler, 2018-11-09:
www.ykhandler.net

major153, 2018-12-10:
Where does it say the version of the camera?   I don't see "4F CN" labeled anywhere.
Mine says YYS.2016.CN on the bottom