shadow-1/yi-hack-v3 #234: Help required to fix the camera

vaidyasr, 2018-11-08:
I have accidentally renamed the /bin/busybox to /bin/busybox_org to accommodate the new busybox, after that I can't rename back to original. It says file system full and can't do any operations. After tried many options, I thought of rebooting may fix the issue. But the camera stays orange after reboot and don't have a idea how to fix it?. Could some one please help?

shadow-1, 2019-01-01:
@vaidyasr 
Unfortunately you have killed your rootfs partition.
To recover from this, flash the stock firmware using the recovery images I linked to on the home page.
Then flash back to yi-hack-v3 firmware again. Everything will be back to normal afterwards.

vaidyasr, 2019-01-09:
@shadow-1 worked perfectly. Thank you.

closed 2019-01-09 by vaidyasr