shadow-1/yi-hack-v3 #235: Camera movement detection to PUSH notification received on phone LONG DELAY

TDPFlorida, 2018-11-12:
I originally purchased (and used) 4 Chinese version Yi Home Smart 720 cameras several years ago and the PUSH notifications took maybe 5-10 seconds (time between moving in front of the camera to a PUSH notification being received on the phone.) 

I decided to upgrade to ENGLISH version cameras last month and bought 4 YI 1080P Home Camera (USA) and a YI Dome 1080 and installed them all. Now the PUSH notification take more than 30 seconds (many times >10-20 minutes) from when I walk in front of the camera. 

This is borderline unacceptable. Example this morning, I drove half-way to work (about 20 minutes) before I received a PUSH notification from the garage. My car that I drove to work with originated from that same garage. I confirmed this because when I look at the time on the phone and the time listed in the notification, it was 20 minutes difference. I have toyed around with the sensitivity and frequency and have gone from not receiving PUSH notifications at all (and I walked and jumped in front of camera) to receiving PUSH notifications all throughout the night when bugs fly in front of the camera with the SAME settings. 

In BOTH cases, the cameras are operating in the eastern US. In addition, the new 1080 English cameras, the firmware has been upgraded to the latest. 2.1.0.0E_201809191630home_y20m for the Home 1080 and 1.9.2.0H_201809031504home_h20m for the 1080 Dome and the Yi Smart Home 720's are on a modified 1.8.6.1B (That includes ENGLISH prompts.) 

Has anyone heard of this or do you think the hack could somehow be doing this? I'm going to remove the hack off one and see if that impacts it? What do you suggest I try? 

Thanks

mfizz1, 2018-11-26:
This is a phone issue for push notifications. I managed to solve it when I was using a Samsung Galaxy phone. I found that if at any point you have ever used ANY power saving features it will automatically put it into saving mode. The only time it would then alert you the next time is if there is a change in network type, i.e. from wifi to mobile network / vpn etc. To ensure this doesn't happen you need to add yi home app to apps which are not monitored or put to hibernate under any case, via settings

TDPFlorida, 2018-11-26:
Thanks...do you know how I do that on an iPhone?
Thanks

TDPFlorida, 2018-11-26:
I'm not sure WHERE to find that setting? I will say that I can walk in front of a camera in my living room and NOT move OR do anything that might cause me to switch networks and the same 20+ second delay happens. On the Chinese version of the camera (which uses MiHome NOT YiHome) the delay is MAYBE 10 seconds. So MiHome vs YiHome may have something to do with a longer delay?
Thanks

mfizz1, 2018-11-27:
I have an android phone and that is what was happening on mine overnight my notifications were really delayed. You could have a check on the app and see if the ping is working correctly on the camera and the servers it connects to. Unfortunately i believe iphone manages its own power settings with no user control. As for 20 seconds delay, I wouldn't say thats out of the question, considering it has to upload the 6 second video clip process it and send it to your phone, i'd say that is still workable. I am more concerned with  your issue of 20-30 minutes