shadow-1/yi-hack-v3 #237: Upgrade to human detection

lucaventurini, 2018-12-02:
Hello,

I have an old 720p home camera. I see that version 2 of the camera have a more sophisticated human detection feature. Do you know if it is possible to get the same with some hacking, or if it is possible to code some program in the camera to do it?

shadow-1, 2019-01-01:
@lucaventurini 
Unlikely the feature can be 'unlocked' so to speak to make it available on these cameras. The features do not appear to exist on the camera itself.

If you forgo official Xiaomi smartphone app support and we develop an alternative way of getting the video stream (for instance RTSP), it is possible to achieve this by coding something yourself or perhaps getting your DVR/Home Automation software (Home Assistant appears to be the popular choice) to implement these features on a central server rather than on the camera itself.

lucaventurini, 2019-01-07:
@shadow-1 Your suggestion is very interesting, but losing the official app would be a bad con. Also, do you think I would be able to access somehow the 6-second videos when movement is recognized? Because if not I'd have to analyze continuously the video stream with some neural network and that would cost a lot of power I fear.

bartlettmic, 2019-01-07:
@lucaventurini after following the instructions in the README of this repo and installing the custom firmware, you should be able to access the camera's filesystem either via SSH or FTP. If you tell the camera to store in the microSD card in the app, you can access those files on the SD card whenever motion is detected. I have a 1080p Dome Black model, and I'm able to find the generated .mp4's in this fashion using SSH, though the directories should be the same over FTP:

```bash
Welcome to HiLinux.
~ # cd /tmp/sd/record/
/tmp/sd/record # ls
2019Y01M07D01H  2019Y01M07D09H  2019Y01M07D11H  2019Y01M07D13H  2019Y01M07D15H  2019Y01M07D17H  2019Y01M07D19H
2019Y01M07D08H  2019Y01M07D10H  2019Y01M07D12H  2019Y01M07D14H  2019Y01M07D16H  2019Y01M07D18H  tmp.mp4.tmp
/tmp/sd/record # cd 2019Y01M07D18H/
/tmp/sd/record/2019Y01M07D18H # ls
02M07S.mp4  09M00S.mp4  13M00S.mp4  17M00S.mp4  22M00S.mp4  26M00S.mp4  30M00S.mp4  36M22S.mp4  40M00S.mp4  44M00S.mp4  48M00S.mp4  52M15S.mp4  56M00S.mp4
03M00S.mp4  10M00S.mp4  14M00S.mp4  19M49S.mp4  23M00S.mp4  27M00S.mp4  31M00S.mp4  37M00S.mp4  41M00S.mp4  45M00S.mp4  49M00S.mp4  53M00S.mp4  57M21S.mp4
07M23S.mp4  11M00S.mp4  15M00S.mp4  20M00S.mp4  24M16S.mp4  28M00S.mp4  33M15S.mp4  38M00S.mp4  42M00S.mp4  46M00S.mp4  50M00S.mp4  54M00S.mp4  58M00S.mp4
08M00S.mp4  12M00S.mp4  16M00S.mp4  21M00S.mp4  25M00S.mp4  29M00S.mp4  34M00S.mp4  39M00S.mp4  43M00S.mp4  47M00S.mp4  51M00S.mp4  55M00S.mp4  59M00S.mp4
```

Until more precise/programmatic means of controlling the camera's hardware are developed (please follow [TheCrypt0](https://github.com/TheCrypt0) and watch for his development of `yi-hack-v4` if you are interested) you can set up some regular task (e.g. crontab) on another device to pull these .mp4's and feed them into a neural network or something of that sort.

lucaventurini, 2019-01-08:
Thank you @bartlettmic that's very interesting, I'll follow the developments!

closed 2019-01-08 by lucaventurini