shadow-1/yi-hack-v3 #242: Yi Cloud Dome 1080p dump in progress :)

BlueBlueBlob, 2018-12-31:
[image]

Hi all,
I'm connected.

But I mess with the home partition before, so I'm waiting my second camera.
I will send dump in the week :)

See you later all

BlueBlueBlob, 2018-12-31:
Or I will compile a custom firmware.
@shadow-1 : are your sdk links still up ?

shadow-1, 2018-12-31:
@BlueBlueBlob 
Yes the SDK's I hosted should still be up (link on main page). I never removed them.

mitchross, 2019-01-03:
Hi @BlueBlueBlob Any update?

BlueBlueBlob, 2019-01-03:
I'm waiting delivery.
Today I will try to provide rootfs and home for y19 (its code name)

phoxy84, 2019-01-03:
Sounds great, can't wait to use my camera. Now I only get the message that the camera does not work outside China :'(

BlueBlueBlob, 2019-01-03:
Well... Next update in 1 week... I bought the Dome 1080p in white but it's not the Cloud 😂
I'm waiting the new one :/

cautroi01, 2019-01-03:
This is stock firmware from YI Technical Support for Yi Cloud Dome 1080p
https://mega.nz/#!N0BHQAoY!XR5R-YI7NutOHSOr8ZL1sSPLeB7ipZP_pcPD5fAqQXg

BlueBlueBlob, 2019-01-03:
> This is stock firmware from YI Technical Support for Yi Cloud Dome 1080p
> https://mega.nz/#!N0BHQAoY!XR5R-YI7NutOHSOr8ZL1sSPLeB7ipZP_pcPD5fAqQXg

Thx but renaming with home_y19 is not wworking. Did they say how to apply ?

cautroi01, 2019-01-03:
The firmware working for me
They say:

1. the firmware file for you to update the camera (do not unzip or rename the firmware, just copy it to SD card root)

If the issue still exists after that, then please help us take the camera firmware logs:
I. insert a Micro SD card (8g-32g,FAT32,completely formatted) into the camera before power on
II. keeps the camera running until the pairing issue occurs once again
III. wait 5 minutes before removing the SD card from camera
IV. copy all the logs and the zbar.raw under SD card via card-reader or laptop and send the files for us to check 
( for example, the files will come in such format:
log_SPM6EEMEL7N94JFM111A_tnp_init_finish.tar.gz
log_SPM6EEMEL7N94JFM111A_yi_login.tar.gz)

BlueBlueBlob, 2019-01-04:
> The firmware working for me
> They say:
>
> 1. the firmware file for you to update the camera (do not unzip or rename the firmware, just copy it to SD card root)
>
> If the issue still exists after that, [...]

Thx a lot :)

BlueBlueBlob, 2019-01-04:
Ok I'm connected, but I can't enter command. I only see Yi process output even pressing enter.

phoxy84, 2019-01-04:
Will this work for both the Chinese and International version? I have Chinese version, can you test with that version as well? If I can help, let me know, allthough I have limited knowledge ;)

TheCrypt0, 2019-01-04:
The image linked by @cautroi01 isn't a firmware image. The suffix "m" means that it's an encrypted update, not a recovery image.

BlueBlueBlob, 2019-01-04:
Yes but I was able to flash it.
@TheCrypt0 : Do I have to paint my camera before to have shell cmd on ttl ?

BlueBlueBlob, 2019-01-04:
After inserting sdcard

himci: card connected!
himci: mmc1: valid phase shift [5, 3] Final Phase 0
mmc1: new ultra high speed SDR104 SDHC card at address aaaa
mmcblk0: mmc1:aaaa SC32G 29.7 GiB
 mmcblk0: p1
stExposureAttr.stAuto.u8Compensation=60
killall: recbackup: no process killed
umount: can't forcibly umount /tmp/cifs: No such file or directory
stExposureAttr.stAuto.u8Compensation=60
p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)
umount: can't remount tmpfs read-only
umount: /dev/mtdblock5 busy - remounted read-only
umount: can't remount tmpfs read-only
stExposureAttr.stAuto.u8Compensation=58
can't run '/sbin/swapoff': No such file or directory
The system is going down NOW!
MMB LEAK(pid=1006): 0x83F46000, 12288 bytes, ''
mmz_userdev_release: mmb<0x83F46000> mapped to userspace 0xb6fca000 will be force unmaped!
Sent SIGTERM to all processes
MMB LEAK(pid=965): 0x835A5000, 4096 bytes, 'AENC(0)StrmBuf'
mmz_userdev_release: mmb<0x835A5000> mapped to userspace 0xb6f09000 will be force unmaped!
MMB LEAK(pid=965): 0x835A6000, 32768 bytes, 'AENC(0) CirBuf'
mmz_userdev_release: mmb<0x835A6000> mapped to userspace 0xb6d62000 will be force unmaped!
MMB LEAK(pid=965): 0x835B3000, 16384 bytes, 'AO(0,0) CirBuf'
mmz_userdev_release: mmb<0x835B3000> mapped to userspace 0xb6c5e000 will be force unmaped!
MMB LEAK(pid=1006): 0x83F49000, 4096 bytes, 'CIPHER_BufIn'
mmz_userdev_release: mmb<0x83F49000> mapped to userspace 0xb6fc9000 will be force unmaped!
MMB LEAK(pid=1006): 0x83F4A000, 4096 bytes, 'CIPHER_BufOut'
mmz_userdev_release: mmb<0x83F4A000> mapped to userspace 0xb6ece000 will be force unmaped!
unlink cmd rsp urb
Sent SIGKILL to all processes
Requesting system reboot

********reboot by watchdog********
Restarting system.

System startup
=========1
hello, open console

 do_auto_sd_update: detect flash size: 0x1000000, gsdexist=1
Interface:  MMC
  Device 0: Vendor: Man 035344 Snr 4a7710f6 Rev: 8.0 Prod: SC32G
            Type: Removable Hard Disk
            Capacity: 30436.5 MB = 29.7 GB (62333952 x 512)
Partition 1: Filesystem: FAT32 "NO NAME    "
   245760   zbar.raw
    36864   log_wifi_connected.tar.gz
    40448   log_first_login.tar.gz
            system volume information/
            record/
            log/
            test/
  1820774   home_y19m

4 file(s), 4 dir(s)

1: zbar.raw!
2: log_wifi_connected.tar.gz!
3: log_first_login.tar.gz!
4: home_y19m!
u-boot_y19 not found!

env_y19 not found!

conf_y19 not found!

kernel_y19 not found!
[... 121 lines cut ...]
^Cusb 1-1: new high-speed USB device number 2 using hiusb-ehci
^C
            _ _ _ _ _ _ _ _ _ _ _ _
            \  _  _   _  _ _ ___
            / /__/ \ |_/
           / __   /  -  _ ___
          / /  / /  / /
  _ _ _ _/ /  /  \_/  \_ ______
___________\___\__________________

[RCS]: /etc/init.d/S00devs
[RCS]: /etc/init.d/S01udev
^C^C^Chimci: mmc1: valid phase shift [5, 3] Final Phase 0
mmc1: new ultra high speed SDR104 SDHC card at address aaaa
mmcblk0: mmc1:aaaa SC32G 29.7 GiB
Not recognise ACTION:change
Not recognise ACTION:change
 mmcblk0: p1
^C^C^C^Crm: can't remove '/sbin/reboot': No such file or directory
exFAT: Version 1.2.9
^Cvm.dirty_background_ratio = 2
vm.dirty_ratio = 2
vm.dirty_writeback_centisecs = 100
vm.dirty_expire_centisecs = 500
fs.mqueue.msg_max = 256

shadow-1, 2019-01-04:
@BlueBlueBlob 
Boot log does not give much away. 😞 

I might investigate the Yi Outdoor camera firmware which also seems to have disabled terminal access. I have more information on that camera.

BlueBlueBlob, 2019-01-04:
@shadow-1 just a question, regarding boot log and kernel information. The y19 is more h20 or y20 ? Because I'm waiting a second one, so I can flash rootfs from other model but I'm not sure :)

shadow-1, 2019-01-04:
@BlueBlueBlob 
Not sure what you are asking.

Your camera is looking for a y18 firmware file to flash from microSD card.
**I do not recommend renaming one of the other firmware files and flashing it. It can soft brick your camera with no way to recover (because we do not have a dump of the stock firmware to restore).**

TheCrypt0, 2019-01-04:
### Discord Server
I've just created a Discord server with a couple of channels. **[Invite link.](https://discord.gg/8A32Z64)**

---

I'm posting this here too, in case someone wants to join have a easier direct communication. The main informations/discoveries about the cam will be posted on the issue too.

BlueBlueBlob, 2019-01-04:
@shadow-1 : y19 not y18.
rootfs is busy box and init.d script, do you think I can flash the rootfs from h20 ?
Like you suggest for Outdoor camera with another rootfs.
Sorry for my English :/

BlueBlueBlob, 2019-01-06:
Well sorry I tried all the day, but I can not mount dumps :/
During this time I extracted last update from y19 : 1.9.3.0E_201812141519home_y19m

BlueBlueBlob, 2019-01-10:
Dump complete

mecaz, 2019-01-13:
Is there any progress?

BlueBlueBlob, 2019-01-15:
@shadow-1 : do you want stock home_y19 and rootfs_y19 ?

shadow-1, 2019-01-18:
@BlueBlueBlob 
It would be helpful to have these files. I am sure an alternative firmware can be developed if it is similar to the other cameras in the series. 😄

BlueBlueBlob, 2019-01-18:
@shadow-1 : sure. V4 is already working with it.
Sorry, I thought you didn't have time.
https://drive.google.com/folderview?id=18v0H6XF54KEcsUXGfRxgw3eWyk6hjIFG

Nomis9, 2019-02-08:
> @BlueBlueBlob
> It would be helpful to have these files. I am sure an alternative firmware can be developed if it is similar to the other cameras in the series. 😄

Will you create modified firmware for Yi Cloud Dome 1080p?
Thanks for answer

alienatedsec, 2021-04-03:
@shadow-1 @BlueBlueBlob  are you still in possession of those stock home_y19 and rootfs_y19 files?