shadow-1/yi-hack-v3 #246: Decrypt audio / video encrypted by "cloud"

jan666, 2019-01-11:
Has anyone found out how the audio/video thats sent by the "cloud" binary to the cloud service is encrypted, where the key is stored and how to decrypt them?

I want to "emulate" the cloud functionality without using the cloud service.

I found out that the cloud app just 2 HTTP (no HTTPS) PUT requests to the cloud service so I redirected the hostname (via DNS) to my own server and started a simple python http server. That works great, but the videos are encrypted.

mvklingeren, 2019-01-11:
This is awesome :-D

mvklingeren, 2019-01-12:
@jan666 if you share the DNS name change..
And the code, I would like to take a look with Wireshark.

There doesn't seem to be any standard for encrypting h264, although hisilicon does provide hardware encryption > my guess is that they use RSA for their frames.

find / | grep key

Anything?

jan666, 2019-01-12:
The DNS Name is motiondetection-eu.oss-eu-central-1.aliyuncs.com

I dont think wireshark will help here. „cloud“ encrypts it locally (there is a log to standard out)

There is a „pub_key“ in /home/base/tools

mvklingeren, 2019-01-12:
Then we need to hunt the priv key.

shadow-1, 2019-01-12:
@jan666 
Interesting work! When I first created yi-hack-v3, I did observe the video feed communication to the cloud service. At that time I was mainly interested in defeating the region lock.

However since he feed was encrypted, I did not go any further.

jan666, 2019-01-12:
Btw: the app still works as it requests just the same URLs via GET. 

My full plan is:

- use „cloud“ to detect motion and notify my NAS
- NAS gets photo / video via HTTP, decrypts them and pushes the Image to my phone via PushOver
- NAS starts ffmpeg RTSP recording to record a video away from the camera (if its taken away)
- Camera records to SD Card as backup 

Problems:

- I dont know how to decrypt the Video / Photo 
- RTSP Server (Yi-Hack-v4 Not out yet)

Nixellion, 2019-01-13:
@jan666 Oh, cool, so you mean it's already possible to just redirect video stream locally? That is awesome, at least for privacy, haha!

Could you share python server code? How do you redirect it back to the phone app?

jan666, 2019-01-14:
@Nixellion not *stream* - "cloud" handles the 6 second video + photo upload after motion is detected

I use the unmodified https://gist.github.com/jongiddy/b26ce43d9da8eb5e9174f4971a80fd9b (dont use it on the internet - anyone can PUT and GET almost anything)

My phone uses the same DNS server (same LAN or VPN) - dont work from the internet

Nixellion, 2019-01-14:
@jan666 Oh.. Well, what about the video stream? It has to go through some proxy as well. Maybe there's a way to hook into that?

shadow-1, 2019-01-18:
@sordfish 
Others are working on a RTSP implementation. However this is getting the video feed directly from the camera in parallel to the cloud feed.

I am not aware if anyone has managed to view the live feed through the cloud service outside of the Yi apps.

shadow-1, 2019-01-18:
@jan666 
It is possible to route the cloud traffic through a device you control and monitor the traffic (I have done this before I first developed this firmware to gain an understanding on how it worked). The trouble is with the encryption. There is not much you can do with encrypted traffic unless you have a method of decrypting it.

Perhaps there is a way of working out how to get around this via the camera. Like I mentioned above, I did not investigate this too deeply as I confirmed this was not relevant to the region lock.

jan666, 2019-01-18:
@shadow-1 its not the traffic thats encrypted, the files are. The PUT is HTTP, not HTTPS. 

It looks like the camera encrypts the file and send it to the Server. The App fetches the files and decrypt them.

shadow-1, 2019-01-18:
@jan666 
You are probably right. It has been a very long time since I examined the traffic.

Jflick58, 2019-01-31:
Any chance you could drop a sample file you've captured for examination? I would be curious to take a look and see if I could figure out what encryption they are using as I can't imagine it being that secure.