shadow-1/yi-hack-v3 #248: Question: How can I reboot camera every hour?

mitchross, 2019-01-20:
My cameras are degrading and require a reboot every hour for the video feed to work. Right now im using a Windows Task to hit http://192.168.1.X/reboot.html every hour to open up firefox. It works, but requires active PC. I started going down the route of creating a nodejs app to connect to the websockets to issue the reboot command ( since thats what reboot.html does ), however the commands seem to get ignored.

shadow-1, 2019-01-24:
@mitchross 
I would have to try this at some stage. However this should be possible to schedule a task on the camera itself using `cron`. This will have to be setup through the command line.

mitchross, 2019-01-25:
Seems reasonable enough, is there an example I can work from?

nicmorais, 2019-03-23:
Run `crontab -e`
Then, in a new line, put the following:
`0 * * * * /sbin/reboot`
Press Esc, type `:wq` and press enter to exit vi.

nelutzu777, 2022-10-19:
Can you  write the entire procedure for "Reboot every 24h"
I don't know why, but my cameras periodically freeze and I can need restart them from the browser http://x.x.x.x.x/reboot.html. It is not very comfortable to reboot when you have many pieces and they get stuck exactly when I need them.

nicmorais, 2022-10-20:
@nelutzu777 
SSH into your devices and do what I said in my previous reply.

nicmorais, 2022-10-26:
@nelutzu777
Oh..
I don't own such camera model, and those I had been working with had crontab avaliable.
The other thing you could do is to program your computer to automatically connect into your camera by SSH and then run "reboot". Stored passwords aren't a good practice, but this could solve your problem if you really need to find a solution.