shadow-1/yi-hack-v3 #249: disable xiaomi server spying?

imacks, 2019-01-25:
i looked at #246 and #175 already. It seems that the current mods do not prevent the camera from communicating with xiaomi servers, which is a deal-breaker in my use scenario. Instead of hacking up a xiaomi server implement, would it be possible to just disable this feature altogether? I can stream the feeds to my home automation server instead.

dskvr, 2019-01-28:
I am also curious about this. In the original [yi-hack](https://github.com/fritz-smh/yi-hack), as well as in [yi-hack-v2](https://github.com/niclet/yi-hack-v2), the best features were that it didn't send footage back to a Chinese server, by either stripping the functionality altogether or an option to disable it. It does not appear v3 offers this functionality. I really doubt that few people wanting to hack their camera is the kind of person who wants to send all their video to a country notorious for using technology as a trojan horse for spying.

dskvr, 2019-01-30:
@imacks V4 is going to explore this, still alpha though ... https://github.com/TheCrypt0/yi-hack-v4