shadow-1/yi-hack-v3 #251: Yi 1080p Home yi-hack-v3 no web login

morphirr, 2019-01-27:
I have **Yi Home 702** patch worked second time, I also now have a **Yi Home 1080p** I put the 
**rootfs_y20**  **home_y20** and followed this instructions it did seem to flash but it is not letting me login like on the **Yi Home 702**  anyone know what is wrong? Camera is now on **firmware 2.0.0.1A_201612051401** update available to **2.1.0.0E_201809191630** can the firmware still be updated after the yi-hack-v3?

-- | --

mfizz1, 2019-01-30:
Sounds silly what version is your camera? us eu cn?

morphirr, 2019-01-31:
Ok I flashed it again with same Sandisk 32GB & I finally got it working on Yi Home 1080p 48US I had to reset the camera several times as it wasn't pairing and would give a timeout msg, I also plugged it into the router for power and proximity to the router (i have found best to do pairing as close to router as possible) I also got that error 2009 at one stage? I then tried web server which let me login so  ProxyChains-ng was enabled which i disabled then when I tried to pair it that did the trick, Can u now update to latest firmware with both these cams now hacked, all new to me so appereciate???

morphirr, 2019-02-01:
@mfizz1 are you a mod as I see that you have had problems with this exact camera 48US and the patch only shows 47US so i dont know if it is suitable for this camera, can anyone clarify that? Although it is working great so far after attempting to apply the patch several times and finally getting it installed I am still on the patch firmware as I'm not sure if it is safe to upgrade and dont want a brick or any more problems as I only just bought these cameras, but thanks for the hard work the patch is brilliant with its additional features, thanks so much for any assistance

mfizz1, 2019-02-02:
The patch is suppose to work for 48cn . More importantly it's about having the same chipset which all of these cameras do. The SD card plays a big factor in the performance. 

You can upgrade and if it goes wrong you can revert using recovery firmware, but it won't. You have root access until you flash recovery

closed 2019-03-07 by morphirr