shadow-1/yi-hack-v3 #252: [Yi Outdoor] FTP Server Doesn't Display Directory Content

alharthyw, 2019-02-02:
Hi,

I've recently bought a brand spankin' new Yi Outdoor Camera [International].
Of course; I had to use your hack to make living with the camera bearable. Yi has gone Apple on us locking features that they should implement. I'm already running your hack on 3 indoor camera beautifully.

RTSP isn't practical for me because I'm not interested in endless streams that die if the connection get shaky, instead I'm interested in FTP to pull MP4's of motion detection recordings, which are then uploaded to my Google Photos with free infinite storage 😝 
It does a wonderful job cataloging clips too.

I simply followed the instructions:
1. Format the 16GB SD card using FAT32
2. Copied home_h30 and rootfs_h30 from your releases onto the root of the SD card
3. Insert the SD card into the camera and turn it on
4. Let the camera flash orange for a short while until it was ready to pair
5. Paired it and connected it to the Yi app. Works as intended, I can see the live footage in the app.

I can see that the camera has some stored clips (from within the Yi App). I can also see that the SD card is in good health and that a number of megabytes have been consumed by recordings. 

The problem: FTP to the root takes me to a blank "folder"; see below. I've tried the same in windows file explorer.
[image]

I've toggled the settings back and forth but it persists:
[image]

[image]

FTP seems to work but unable to list content. I've tried copying the exact url including /tmp/sd/ from other working FTP Yi cameras [Dome 1080p] in my home but still.
I've tried entering an incorrect pair user/pass and it rejected them as it should.
I'm logging in as "root" and a blank password.

Please help!

drlarsen77, 2019-02-02:
Personally, I turn FTP and Telnet off as both are insecure and I use Dropbear (SSH) instead. I also set the root password to something complex and use SSH keys so I can securely connect without a password.

Running this on my server (one line) will transfer the previous hour's recording to the current folder.
``ssh -C root@192.168.1.23 "cd /tmp/sd/record/`date -u -d '1 hour ago' '+%YY%mM%dD%HH'`; tar cf - ." | tar xvf -``
Note that there are backticks around the date command.

I'm not sure if that helps you or not.

Even if you're running Windows, you can still do this using Windows Subsystem for Linux (WSL) if you're using Windows 10 (or Server 2016+). Or something like Git Bash.

alharthyw, 2019-02-02:
> Personally, I turn FTP and Telnet off as both are insecure and I use Dropbear (SSH) instead. I also set the root password to something complex and use SSH keys so I can securely connect without a [...]

This does sound like a good workaround for my case. That is if ftp remains broken. Although, the endgame is to map all the cameras to network drives via ftp; or an alternative protocol if that would work. Then point Google's "Back up and sync" to them directly; cam to cloud archive.

An update: the ftp failed to list files in the browser and windows explorer. However, using a syncing app "GoodSync", I managed to list the files and back them up to my local hard drive.

alharthyw, 2019-02-04:
> @alharthyw
> I recall seeing this issue in the past. The issue is with Windows Explorer. It appears that Windows Explorer is limiting access to only the home directory. The home directory by default is empty.
> 
> I suggest trying a dedicated FTP application file FileZilla or WinSCP.
> 
> Edit: Just released you found Windows Explorer causing the issue in your second post.

I did find a solution that works for me. But would be great if we could help people ending up on this page.

ftp works great in my Yi Dome cameras. Perhaps it's a small patch to get it running on Yi outdoor? maybe it's not so simple 🤓 

Yi Outdoor = 192.168.1.35
[image]

Yi Dome = 192.168.1.31
[image]
[image]

shadow-1, 2019-03-05:
@alharthyw 
Very interesting.

I suspect it is something very simple that I probably missed in the Yi Outdoor build.

moforlive, 2019-03-18:
Have the same problem with my Yi Home 47US.
1.8.7.0D_201708091510
Firmware 0.1.6