shadow-1/yi-hack-v3 #256: Connected for just 2 minutes

volcane2, 2019-02-10:
First ,thank u for a wonderful hacked,
YiHome 1080p 48CN camera connected for only 2 minutes and been blocked again "Pls check your network/power issue" error.I have done repeating hacked as guide for at least 6-7 times.Its connected beautifully but can only be used for 2-3 minutes and been banned again?

shadow-1, 2019-02-20:
@volcane2 
Hmm, not too sure what the issue is.
Just to confirm, have you have configured ProxyChains-ng and confirmed that it is working?

It appears to be as it is pairing and working for a few minutes.

Another question, are you using the latest Yi Home (Chinese version) app?
I have had similar issues in the past by using a very old version of the Android app.

volcane2, 2019-02-20:
Yeah, thanks for responds, yes, its connected successfully with ProxyChains-ng as described and latest Chinese Yi Home app ,i tried a few times,connected and live view for about 3-4 minutes, and error came out "pls check your network/power" disconnected, i think they managed to trace my ip and banned immediately ! Any suggestion? I can't use old Yi Home Chinese App as i have other YiHome camera in same phone.

littlespazo, 2019-04-18:
Hi Shadow-1 and Volcane2,

I also have a YiHome 1080p 48CN camera and have been facing a similar issue. When i tested it with YiHack v3 0.14beta 2 , sometime last year it worked very well. Then i accidentally, updated the firmware on the Yihome App, and everything went bonkers.

Tried re-installing the camera's with  YiHack v3 0.14beta 2, YiHack v3 0.15, and YiHack v3 0.16, but with no success.
I even tried testing all the various edition of YiHome App from the repository to the newest one on he website, with no luck.

I'm guessing a particular YiHome apk, worked with a particular Hack, e.g. Yihack v3 0.16? Am i right Shadow-1? Any chance you can share which YiHome version you used when testing?