shadow-1/yi-hack-v3 #261: Working proxies

major153, 2019-03-25:
Anyone have a working CN proxy?

I tried about 30 from http://spys.one/free-proxy-list/CN/ and they don't seem to work.  

The setup said it failure & restart after connecting to wifi.  I tried the test proxy feature in ProxyChains-ng and they all give me 3 red crosses

random00f, 2019-03-26:
Hello! I tried everything, and not only from http://spys.one. maybe something wrong doing
[image]
?

random00f, 2019-03-27:
Solution: type of proxy should be entered in lower case. but finding a working proxy CN is very difficult. Hong Kong is not suitable, you need specifically CN