shadow-1/yi-hack-v3 #262: Firmware not upgrading/ No Hack

kpj5425, 2019-03-25:
Factory Firmware Ver. 2.1.0.0E_201809191630

I followed the steps step by step and I don't get a flashing yellow for the approx 30 sec. and can't complete this process so I can't use it in my Home assistant. Though Home Assistant sees the camera... not sure what is going on. Please advise.

junhau89, 2019-03-26:
Hi, mine Yi 1080p since after upgrade from v3 0.1.4 version to v3 0.1.5 version. Then  currently stuck at v3 0.1.5 version. Unable to update to latest version v3 or v4... also unable to recovery back to official version. Any idea to solve it??

nelutzu777, 2019-04-18:
> Factory Firmware Ver. 2.1.0.0E_201809191630

Same here. I can't get the firmware to flash at all.  I have Yi 1080p Home Camera, SN: 6FUS... V1

grantmacnamara, 2019-04-25:
Same here. Yi 1080p Home Camera wont load the firmware. Works fine on my 720 version. 
Tried, reset, reformat, etc
Firmware 4.5.0.0A201902261502

Caobara, 2019-05-16:
I not able to get the firmware to work on my 1080P Dome. IT's seemed that it was working but still getting yellow flashing . Any ideas. I have chosen the correct files formatted fat32 ect...Thanks everyone

smksky, 2019-06-11:
i have saem issue at  YIhome(y27) 1.8.7 201809191630

isbkch, 2020-10-23:
same issue on "YI 1080p Home" 8.2.0.0A_202007211802