shadow-1/yi-hack-v3 #265: Xiaomi 1080P Smart Camera

afoninsky, 2019-04-21:
I have Xiaomi 1080P Smart Camera (like [this one](https://www.aliexpress.com/item/Original-Xiaomi-Mijia-New-1080P-IP-Camera-130-Degree-FOV-Night-Vision-2-4Ghz-Dual-band/32899846324.html)).

Can I use one of firmwares? And what firmware should I use? Thx

nicmorais, 2019-04-25:
Hello
Please tell us the first four digits of the QR code in the back of camera.

afoninsky, 2019-04-26:
Hello.

Sorry it's mounted on the wall and I don't want to unmount without reasons.
If you want to know exact model - it is "Mi Basic 1080P". Otherwise tell me and I'll unmount it.

nicmorais, 2019-04-26:
Unfortunately, this hack does not support your camera, neither yi-hack-v4 does.

nicmorais, 2019-04-27:
@afoninsky but, if the developers get what cpu, ram and rom this device has, and a firmware dump, they can work to support your camera

afoninsky, 2019-04-27:
I'll reopen issue then :) If developers contact me I'll try to provide information as far as I can.

nicmorais, 2019-04-27:
@afoninsky take a look at this: [Filipowicz251/mijia-1080P-hacks](https://github.com/Filipowicz251/mijia-1080P-hacks)

afoninsky, 2019-04-29:
Yep I got it thx. Kind of tricky but looks like it's the only way for now :(
https://github.com/Filipowicz251/mijia-1080P-hacks/issues/55

closed 2019-04-29 by afoninsky