shadow-1/yi-hack-v3 #266: Download Proxy List

mptacnik, 2019-04-21:
when i try to download the proxy list i get this:

wget -O /tmp/temp_proxy_list.txt https://freevpn.ninja/free-proxy/txt 

Connecting to freevpn.ninja (185.53.179.7:443)
wget: server returned error: HTTP/1.1 403 Forbidden

Return Code: 1

nicmorais, 2019-04-25:
This website is down.

mptacnik, 2019-04-26:
temporarily or for good?

nicmorais, 2019-04-26:
I think it's gone.
Also yi-hack-v3 is not being maintained anymore, but [yi-hack-v4](https://github.com/TheCrypt0/yi-hack-v4) is.

stonelf, 2019-09-17:
try  https://www.kuaidaili.com/free/inha/2/?_t=t

stonelf, 2019-09-17:
now you have to find a few proxy(http/socks) servers for yourself, and fill them in the **ProxyChains-ng Configuration** page, save and then test to make sure they work.

gam2046, 2019-10-02:
> temporarily or for good?

It seems to be permanently closed