shadow-1/yi-hack-v3 #268: yi 1080p home camera 6FUS

ale953, 2019-05-15:
I have bought a yi 1080p home camera 6FUS, and I want to know if it's possible to hack it with a new firmware.

maelboutin, 2019-07-20:
Same here, any info  ?

tomascrespo, 2019-07-29:
Same here. 
In project description says compatible with _YI Home 1080p_, and this is a Yi Home 1080p. 
Are Yi Home 1080p with different codes (6FUS, ...)? So... with what codes is yi-hack-v3 compatibles?

I tried this hack with my 6FUS and I have not been able to get it working.

Thanks for wonderful job!

ykhandler, 2019-07-29:
6FUS use different chipset, you need to be patient for the developer to create it. Btw does the firmware version of 6FUS is 
4.5.0.0A_201902261502 ?
Like 6FCN?

pidiman, 2019-08-03:
Hello. I have Yi home 1080p - 6FUSY2183J and I tried every options but every firmvares doesnt worked.

Everytime when I upload firmvare to sd card and plug in to camera... camera after 5-10seconds says: welcome to yi home.. waiting to connect... 

Any helps ? thanks

christiandulle, 2019-08-11:
same problem here. also with 6fus ... any ideas how to use the yi-hack?

15goudreau, 2019-09-07:
is anyone still working on this?

steef84, 2019-09-27:
see here https://github.com/roleoroleo/yi-hack-6FUS_4.5.0

xfceKris, 2020-03-16:
I know this is an older Issue report, but I have four 6FUS cameras, all running stock firmware version 2.1.0.0E. Is there any support for these models? I've looked through V2, V3, V4, and MStar issue logs and can't find anyone with the same model.