shadow-1/yi-hack-v3 #273: Xiaomi Mijia IMILAB Xiaobai H.265 1080P Smart Home IP Camera 360°

errellion, 2020-04-07:
Hello,

sorry if I'm writing nonsense here, but I can't identify if camera model - Xiaomi Mijia IMILAB Xiaobai H.265 1080P Smart Home IP Camera 360°  (https://www.banggood.com/International-Version-Xiaomi-Mijia-IMILAB-Xiaobai-H_265-1080P-Smart-Home-IP-Camera-360-PTZ-AI-Detection-WIFI-Security-Monitor-from-Xiaomi-Eco-system-p-1450083.html?rmmds=myorder&cur_warehouse=GWTR)

can it be same like "Yi 1080p Dome"?

I even tried to flash it with 2GB SD card FAT32 formatted with files home_h20 and rootfs_h20 on it, but yellow light is not flashing, then it changes to blinking blue and camera boots up normally (can be viewed in an app only).

Camera has probably old stock firmware 1.6.xx something.

sdoup, 2020-10-04:
i have the same cam and looking the answre.didi you manage to make it work?>>

supermemo007, 2021-01-25:
Hello did you get any update on this? I have the same model too! Any chance to get the streaming working on third party application?